jrl-umi3218/mc_rtc #4: error: ‘class Eigen::LSSOL_QP’ has no member named ‘persistence’

xeroxx, 2020-01-17:
Hi,
I've got this error after trying to install the latest version of mc_rtc on a new machine.
I just wanted to report it.

```
[  4%] Building CXX object src/CMakeFiles/Tasks.dir/LSSOLQPSolver.cpp.o
cd /home/saeed/devel/src/Tasks/build/src && /usr/bin/c++  -DBOOST_ALL_NO_LIB -DBOOST_CHRONO_DYN_LINK -DBOOST_SERIALIZATION_DYN_LINK -DBOOST_TIMER_DYN_LINK -DLSSOL_SOLVER_FOUND -DTasks_EXPORTS -I/home/saeed/devel/src/Tasks/build -I/home/saeed/devel/src/Tasks/build/include -I/home/saeed/devel/src/Tasks/include -I/home/saeed/devel/src/Tasks/src -I/home/saeed/devel/src/Tasks/build/src/../include -isystem /usr/local/include/eigen3  -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  -Wall -Wextra -pedantic -O2 -g -DNDEBUG -fPIC   -mfpmath=sse -msse2 -Dtasks_EXPORTS -std=gnu++11 -o CMakeFiles/Tasks.dir/LSSOLQPSolver.cpp.o -c /home/saeed/devel/src/Tasks/src/LSSOLQPSolver.cpp
[  4%] Built target copy-sources-12-tasks-python-bindings
/home/saeed/devel/src/Tasks/src/LSSOLQPSolver.cpp: In constructor ‘tasks::qp::LSSOLQPSolver::LSSOLQPSolver()’:
/home/saeed/devel/src/Tasks/src/LSSOLQPSolver.cpp:25:10: error: ‘class Eigen::LSSOL_QP’ has no member named ‘persistence’
   lssol_.persistence(true);
          ^
src/CMakeFiles/Tasks.dir/build.make:195: recipe for target 'src/CMakeFiles/Tasks.dir/LSSOLQPSolver.cpp.o' failed
make[2]: *** [src/CMakeFiles/Tasks.dir/LSSOLQPSolver.cpp.o] Error 1
make[2]: Leaving directory '/home/saeed/devel/src/Tasks/build'
CMakeFiles/Makefile2:1364: recipe for target 'src/CMakeFiles/Tasks.dir/all' failed
make[1]: *** [src/CMakeFiles/Tasks.dir/all] Error 2
make[1]: Leaving directory '/home/saeed/devel/src/Tasks/build'
Makefile:143: recipe for target 'all' failed
make: *** [all] Error 2
Build failed for Tasks
```

xeroxx, 2020-01-17:
I commented out these lines in Tasks/CMakeLists.txt and the compilation was successful after that.

```
find_package(eigen-lssol QUIET)
if(${eigen-lssol_FOUND})
  add_project_dependency(eigen-lssol REQUIRED)
endif()
```

arntanguy, 2020-01-21:
Hi,
This change was introduced 2 months ago in `eigen-lssol` to fix a typo `persistance` -> `persistence` in the function name.  However `eigen-lssol` is not part of the public repositories, so you'll have to build against the publicly supported `qld` solver (which you did by disabling eigen-lssol in your previous message).

gergondet, 2020-01-31:
Yes. The issue is caused by an outdated eigen-lssol install. Closing this

closed 2020-01-31 by gergondet