jrl-umi3218/mc_rtc #6: Issue while adding a new robot to mc_rtc

mohamadmahdavian, 2020-01-28:
Hi

I am trying to add our robot to mc_rtc package and I am getting following error while making the mc_rtc package. I have added a folder named sm_r_v1_description to the mc_rtc_data folder which contains the model and .urdf files. I have attached our robot.cpp and robot.h and CMakeFile.txt files. So you can check them.

[my_files.zip](https://github.com/jrl-umi3218/mc_rtc/files/4120327/my_files.zip)

Thank You

```
Building CXX object tests/CMakeFiles/test_mc_rbdyn.dir/test_mc_rbdyn.cpp.o
cd /home/saeed/repos/3rd_party/lipm/mc_rtc/build/tests && /usr/bin/c++   -DBOOST_TEST_DYN_LINK -DBOOST_TEST_MAIN -I/home/saeed/repos/3rd_party/lipm/mc_rtc/build -I/home/saeed/repos/3rd_party/lipm/mc_rtc/build/include -I/home/saeed/repos/3rd_party/lipm/mc_rtc/include -I/home/saeed/repos/3rd_party/lipm/mc_rtc/build/tests/include -I/home/saeed/repos/3rd_party/lipm/mc_rtc/src/../include -isystem /usr/local/include -isystem /usr/include/eigen3  -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  -fvisibility=hidden   -mfpmath=sse -msse2 -std=gnu++11 -o CMakeFiles/test_mc_rbdyn.dir/test_mc_rbdyn.cpp.o -c /home/saeed/repos/3rd_party/lipm/mc_rtc/tests/test_mc_rbdyn.cpp
/home/saeed/repos/3rd_party/lipm/mc_rtc/src/mc_robots/sm_r_v1.cpp:19:1: error: ‘sm_r_v1RobotModule’ does not name a type
 sm_r_v1RobotModule::sm_r_v1RobotModule(bool fixed) : mc_rbdyn::RobotModule(SM_R_V1_DESCRIPTION_PATH, "sm_r_v1")
 ^
src/mc_robots/CMakeFiles/sm_r_v1.dir/build.make:65: recipe for target 'src/mc_robots/CMakeFiles/sm_r_v1.dir/sm_r_v1.cpp.o' failed
make[2]: *** [src/mc_robots/CMakeFiles/sm_r_v1.dir/sm_r_v1.cpp.o] Error 1
make[2]: Leaving directory '/home/saeed/repos/3rd_party/lipm/mc_rtc/build'
CMakeFiles/Makefile2:2194: recipe for target 'src/mc_robots/CMakeFiles/sm_r_v1.dir/all' failed
make[1]: *** [src/mc_robots/CMakeFiles/sm_r_v1.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

arntanguy, 2020-01-28:
This is simply a namespace issue. In you header file, you've decalred:

```cpp
namespace mc_robots
{
struct MC_ROBOTS_DLLAPI sm_r_v1RobotModule : public mc_rbdyn::RobotModule
{
...
};
}
```

While in your C++ file you're implementing
```cpp
sm_r_v1RobotModule::sm_r_v1RobotModule(bool fixed) : c_rbdyn::RobotModule(SM_R_V1_DESCRIPTION_PATH, "sm_r_v1")
```

The namespace `sm_r_v1RobotModule::` should be replaced by `mc_robots::` (or the opposite depending on your preference ;)

mohamadmahdavian, 2020-01-29:
Yes. You were right. Thank You for your help

closed 2020-01-29 by mohamadmahdavian