jrl-umi3218/mc_rtc #7: checking linker flags

fkanehiro, 2020-01-28:
https://github.com/jrl-umi3218/mc_rtc/blob/master/plugins/ROS/CMakeLists.txt#L30-L35 is checking the existence of a linker flag using `check_cxx_compiler_flag()` but it doesn't work.
Is there a method to check linker flags?

arntanguy, 2020-01-28:
This is something declared in the [jrl-cmakemodules](https://github.com/jrl-umi3218/jrl-cmakemodules), which are included in `mc_rtc` as a submodule.

Are you submodules up to date? You can update them with
```
git submodule sync
git submodule update
```

And then try running `cmake` again.

fkanehiro, 2020-01-28:
I think this is not a matter of jrl-cmakemodules.
`check_cxx_compiler_flag()` is a function defined in cmake and it is for checking compiler flags.

gergondet, 2020-01-31:
What is the issue this is creating for you? As far as I know, there is no way to check for linker flags but the way the compiler flag checking works is just to check that `${COMPILER} ${FLAG} trivial.cpp` does not complain about unknown flags.

This flag only makes sure that mc_tasks_ros is linked to the ROS plugin so that when the plugin is loaded the ROS-related tasks are also loadable from JSON/YAML without requiring the user to link with mc_tasks_ros explicitly. Thinking about it, the better solution would be to make sure the linking happens for good reasons. I'll try to get a solution for this next week.

fkanehiro, 2020-01-31:
I'm trying to use ROS + mc_rtc on macOS.
As this code doesn't check if linker flags are supported or not, cmake always add linker flags `--as-needed` and `--no-as-needed`.  But ld on macOS doesn't support these linker flags and it fails.

PR #9 (2020-02-03): [ROS] Remove the need for --no-as-needed flag
Closes #7
commit 3b11c63c (2020-02-03): [ROS] Remove the need for --no-as-needed flag

Closes #7
commit 6dfeba5a (2020-02-03): [ROS] Remove the need for --no-as-needed flag

Closes #7
closed 2020-02-03 by gergondet