jrl-umi3218/mc_rtc #11: Requested creation of object named ROS which has not been loaded

fkanehiro, 2020-02-05:
I tried to run `choreonoid sim_mc.cnoid --start-simulation` and choreonoid died with an error message `Requested creation of object named ROS which has not been loaded`.
What does this error message mean?

```
$  choreonoid sim_mc.cnoid --start-simulation
PDcontroller0: onInitialize()
Loading body customizer "/home/kanehiro/openrtp/share/OpenHRP-3.1/customizer/HRP5PCustomizer.so" for HRP5P
Loading body customizer "/home/kanehiro/openrtp/share/OpenHRP-3.1/customizer/HRP2Customizer.so" for HRP2, HRP2NF1, HRP2TIG, HRP2D, HRP2JRL, HRP2A, HRP2JSK, HRP2JSKNT, HRP2W, HRP2K, HRP2SH, HRP2DRC, HRP2JVRC, HRP2KAI, HRP3, HRP4L, HRP4, HRP4Cg, HRP4R, HRP4J
PDcontroller0: on Activated
[PDcontroller0] Gain file [PDgains_sim.dat] opened
[PDcontroller0] tlimit_ratio is set to 1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1
[PDcontroller0] loadGain
[PDcontroller0]   R_HIP_P, pgain = 5000, dgain = 30
[PDcontroller0]   R_HIP_R, pgain = 5000, dgain = 30
[PDcontroller0]   R_HIP_Y, pgain = 5000, dgain = 30
[PDcontroller0]   R_KNEE, pgain = 6000, dgain = 50
[PDcontroller0]   R_ANKLE_R, pgain = 3000, dgain = 20
[PDcontroller0]   R_ANKLE_P, pgain = 3000, dgain = 20
[PDcontroller0]   L_HIP_P, pgain = 5000, dgain = 30
[PDcontroller0]   L_HIP_R, pgain = 5000, dgain = 30
[PDcontroller0]   L_HIP_Y, pgain = 5000, dgain = 30
[PDcontroller0]   L_KNEE, pgain = 6000, dgain = 50
[PDcontroller0]   L_ANKLE_R, pgain = 3000, dgain = 20
[PDcontroller0]   L_ANKLE_P, pgain = 3000, dgain = 20
[PDcontroller0]   WAIST_Y, pgain = 5000, dgain = 32
[PDcontroller0]   WAIST_P, pgain = 5000, dgain = 32
[PDcontroller0]   WAIST_R, pgain = 5000, dgain = 32
[PDcontroller0]   NECK_Y, pgain = 1000, dgain = 21
[PDcontroller0]   NECK_R, pgain = 1000, dgain = 21
[PDcontroller0]   NECK_P, pgain = 1000, dgain = 21
[PDcontroller0]   R_SHOULDER_P, pgain = 1100, dgain = 70
[PDcontroller0]   R_SHOULDER_R, pgain = 1100, dgain = 70
[PDcontroller0]   R_SHOULDER_Y, pgain = 1100, dgain = 70
[PDcontroller0]   R_ELBOW_P, pgain = 1200, dgain = 80
[PDcontroller0]   R_ELBOW_Y, pgain = 1200, dgain = 80
[PDcontroller0]   R_WRIST_R, pgain = 500, dgain = 24
[PDcontroller0]   R_WRIST_Y, pgain = 500, dgain = 24
[PDcontroller0]   R_UTHUMB, pgain = 30, dgain = 3
[PDcontroller0]   R_LTHUMB, pgain = 30, dgain = 3
[PDcontroller0]   R_UINDEX, pgain = 30, dgain = 3
[PDcontroller0]   R_LINDEX, pgain = 30, dgain = 3
[PDcontroller0]   R_ULITTLE, pgain = 30, dgain = 3
[PDcontroller0]   R_LLITTLE, pgain = 30, dgain = 3
[PDcontroller0]   L_SHOULDER_P, pgain = 1100, dgain = 70
[PDcontroller0]   L_SHOULDER_R, pgain = 1100, dgain = 70
[PDcontroller0]   L_SHOULDER_Y, pgain = 1100, dgain = 70
[PDcontroller0]   L_ELBOW_P, pgain = 1200, dgain = 80
[PDcontroller0]   L_ELBOW_Y, pgain = 1200, dgain = 80
[PDcontroller0]   L_WRIST_R, pgain = 500, dgain = 24
[PDcontroller0]   L_WRIST_Y, pgain = 500, dgain = 24
[PDcontroller0]   L_UTHUMB, pgain = 30, dgain = 3
[PDcontroller0]   L_LTHUMB, pgain = 30, dgain = 3
[PDcontroller0]   L_UINDEX, pgain = 30, dgain = 3
[PDcontroller0]   L_LINDEX, pgain = 30, dgain = 3
[PDcontroller0]   L_ULITTLE, pgain = 30, dgain = 3
[PDcontroller0]   L_LLITTLE, pgain = 30, dgain = 3
[StateHolder0] onInitialize()
StateHolder: dt = 0.005
[DataLogger0] onInitialize()
Loading default global configuration /home/kanehiro/jrl_new/etc/mc_rtc.yaml
Enabled plugins: ROS (autoload)
GUI server enabled
Will serve data on:
- ipc:///tmp/mc_rtc_pub.ipc
- tcp://*:4242
Will handle requests on:
- ipc:///tmp/mc_rtc_rep.ipc
- tcp://*:4343
Loading additional plugin configuration: "/home/kanehiro/jrl_new/lib/mc_plugins/etc/ROS.yaml"
Requested creation of object named ROS which has not been loaded
Segmentation fault (コアダンプ)
```

gergondet, 2020-02-05:
It would indicate the ROS plugin failed to load. I am not sure this should cause a segfault. Can you trace it?

Can you run controllers with mc_rtc_ticker?

fkanehiro, 2020-02-05:
Here is the trace and I can run mc_rtc_ticker.

```

Requested creation of object named ROS which has not been loaded

Thread 19 "choreonoid" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff8d76e700 (LWP 1005)]
0x00007fffc3fa2841 in POA_OpenHRP::MCControlService::~MCControlService (
    this=0x7fff84091990,
    __vtt_parm=0x7fff7efea268 <VTT for OpenHRP::MCControlServiceSVC_impl+40>,
    __in_chrg=<optimized out>)
    at /home/kanehiro/src/hrpsys-humanoid/build/idl/MCControlServiceSk.cpp:1039
1039    POA_OpenHRP::MCControlService::~MCControlService() {}
(gdb) bt
#0  0x00007fffc3fa2841 in POA_OpenHRP::MCControlService::~MCControlService (
    this=0x7fff84091990,
    __vtt_parm=0x7fff7efea268 <VTT for OpenHRP::MCControlServiceSVC_impl+40>,
    __in_chrg=<optimized out>)
    at /home/kanehiro/src/hrpsys-humanoid/build/idl/MCControlServiceSk.cpp:1039
#1  0x00007fff7edcb5bc in OpenHRP::MCControlServiceSVC_impl::~MCControlServiceSVC_impl (this=0x7fff84091990, __in_chrg=<optimized out>,
    __vtt_parm=<optimized out>)
    at /home/kanehiro/jrl_new/src/mc_openrtm/src/MCControlServiceSVC_impl.cpp:25
#2  0x00007fff7edce1ce in MCControl::MCControl (
    this=this@entry=0x7fff840871e0, manager=manager@entry=0xd6f3c0,
    __in_chrg=<optimized out>, __vtt_parm=<optimized out>)
    at /home/kanehiro/jrl_new/src/mc_openrtm/src/MCControl.cpp:87
#3  0x00007fff7edd9c41 in RTC::Create<MCControl> (manager=0xd6f3c0)
    at /home/kanehiro/openrtp/include/openrtm-1.1/rtm/RTObject.h:4926
#4  0x00007fffc4887ffa in RTC::FactoryCXX::create (this=0x7fff84066ec0,
    mgr=<optimized out>) at Factory.cpp:103
#5  0x00007fffc4898c17 in RTC::Manager::createComponent (this=0xd6f3c0,
    comp_args=comp_args@entry=0x7fff8405cd60 "MCControl") at Manager.cpp:769
#6  0x00007fffc48be387 in RTM::ManagerServant::create_component (
    this=0xd7fba0, module_name=0x7fff8405cd60 "MCControl")
    at ManagerServant.cpp:335
#7  0x00007fffc49a097c in _0RL_lcfn_f30e1745ea13c2d7_90000000 (
    cd=0x7fff8d76da70, svnt=<optimized out>)
    at ../../../../src/lib/rtm/idl/ManagerSK.cc:469
#8  0x00007fffc5444759 in omniCallHandle::upcall(omniServant*, omniCallDescriptor&) () from /usr/lib/libomniORB4.so.1
#9  0x00007fffc49a7293 in RTM::_impl_Manager::_dispatch (this=<optimized out>,
    _handle=...) at ../../../../src/lib/rtm/idl/ManagerSK.cc:1296
#10 0x00007fffc543d7bd in omni::omniOrbPOA::dispatch(omniCallHandle&, omniLocalIdentity*) () from /usr/lib/libomniORB4.so.1
#11 0x00007fffc541b7a8 in omniLocalIdentity::dispatch(omniCallHandle&) ()
   from /usr/lib/libomniORB4.so.1
#12 0x00007fffc545ebc0 in omni::GIOP_S::handleRequest() ()
   from /usr/lib/libomniORB4.so.1
#13 0x00007fffc545fa48 in omni::GIOP_S::dispatcher() ()
   from /usr/lib/libomniORB4.so.1
#14 0x00007fffc545c3f5 in omni::giopWorker::real_execute() ()
   from /usr/lib/libomniORB4.so.1
#15 0x00007fffc545cb1f in omni::giopWorker::execute() ()
   from /usr/lib/libomniORB4.so.1
#16 0x00007fffc540f56d in omniAsyncWorkerInfo::run() ()
   from /usr/lib/libomniORB4.so.1
#17 0x00007fffc540fbff in omniAsyncWorker::run(void*) ()
   from /usr/lib/libomniORB4.so.1
#18 0x00007fffc514f779 in omni_thread_wrapper ()
   from /usr/lib/libomnithread.so.3
#19 0x00007ffff52d16ba in start_thread (arg=0x7fff8d76e700)
    at pthread_create.c:333
#20 0x00007ffff6d7241d in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
(gdb)
```

gergondet, 2020-02-05:
I get a crash of the MCControl component when I make sure the ROS plugin creation fails. I guess in some situations it might crash choreonoid all-together. I'll try to get mc_rtc to handle this more gracefully but in the meantime we can try to figure out why the plugin fails to load:

1. Check that you have your ROS environment correctly setup when you run choreonoid
2. Assuming this is ok: `LD_DEBUG=libs choreonoid sim_mc.cnoid --start-simulation 2> /tmp/ld.log` then you can check in `/tmp/ld.log` what goes wrong when mc_rtc tries to load the ROS plugin, we can maybe better figure out the issue from there

fkanehiro, 2020-02-05:
ROS environment variables were not set properly and choreonoid was failing to find libroscpp.so.
After setting up ROS environment, it worked well.
Sorry!

closed 2020-02-05 by fkanehiro

gergondet, 2020-02-05:
No problem. Glad we could figure it out. I'll try to investigate the crash