jrl-umi3218/mc_rtc #14: CoMTask

stephane-caron, 2020-03-07:
@Saeed-Mansouri Answering your [mc_rtc-related question about the CoMTask](https://github.com/stephane-caron/lipm_walking_controller/issues/42):

> As you can see below, it seems that this->eval() returns a cross-product of the com error and dimweight.

The corresponding eigen function, ``cwiseProduct()``, is a component-wise product rather than a cross-product. Check out the function documentation for details.

arntanguy, 2020-03-08:
Original issue:

> Hi Dr. Caron,
> 
> I added a new `PositionTask` to control pelvis position in the z-direction to have more natural walking. Also, I changed `dimweight` of CoMTask to `{0.5, 1., 0.}` and `{0., 0., 1.}` was set for the new `PositionTask`. However, I found that the third array of "com" element in the CoMTask of gui didn't change at all. I tried to find that what value returned for "com" element.
> 
> ```cpp
> void CoMTask::addToGUI(mc_rtc::gui::StateBuilder & gui)
> {
>   TrajectoryTaskGeneric<tasks::qp::CoMTask>::addToGUI(gui);
>   gui.addElement({"Tasks", name_},
>                  mc_rtc::gui::Point3D("com_target", [this]() { return this->com(); },
>                                       [this](const Eigen::Vector3d & com) { this->com(com); }),
>                  mc_rtc::gui::Point3D("com", [this]() { return (this->com() - this->eval()).eval(); }));
> }
> ```
> 
> As you can see below, it seems that `this->eval()` returns a cross-product of the com error and `dimweight`. However, I couldn't understand the physical interpretation of this process. I will appreciate it if you could explain it.
> 
> ```cpp
> template<typename T>
> Eigen::VectorXd TrajectoryTaskGeneric<T>::eval() const
> {
>   if(selectorT_)
>   {
>     return selectorT_->eval().cwiseProduct(trajectoryT_->dimWeight());
>   }
>   return errorT->eval().cwiseProduct(trajectoryT_->dimWeight());
> }
> ```

Thanks for reporting an issue. We are solving a weighted QP, where each task has a given weight. In addition to that, you might want a bit more control, and individually control the weights for each dimesion of a task: this is the role of `dimWeight`. For instance you can do:
```cpp
comTask->weight(1000); // comTask has a weight of 1000
comTask->dimWeight({1,1,0.5}; x=1000, y=1000, z=500
```

```cpp
mc_rtc::gui::Point3D("com", [this]() { return (this->com() - this->eval()).eval(); }));
```

I believe the intent here was to avoid the extra cost of computing the `CoM` twice (once in the `Tasks` library, and once in `mc_rtc` wrapper) by subtracting the task error to the current `CoM` target. Without `dimWeight`, this would be true, however you are right, and this implementation is wrong when `dimWeight != Eigen::VectorXD::Ones()`! With this formula, we are only subtracting a percentage of the actual CoM position error, and not the full error.

Below is an example showing the issue with a `CoMTask` with `dimWeight={1,1,0.1}`:
- Dashed red: CoM computed with `robot().com` (calling `rbd::computeCoM(mb, mbc)`
- Blue is the CoM computed with `this->com() - this->eval()).eval()` (which is what we log)  

On the right side, I increased the weight of the posture task to ensure that the CoM task would not converge to its target.

[image]

We can come up with a simple fix that:
- Exposes the computed CoM from `Tasks::CoMTask`
- Uses it in `mc_tasks::CoMTask` wrapper instead of the current implementation

stephane-caron, 2020-03-08:
Nice catch!

PR #15 (2020-03-09): [CoMTask] Fix computation of logged CoM position with dimWeight
Fixes #14: Fixes a bug in logging/gui of the com position when using `dimWeight`.
Requires the corresponding PR in `Tasks`: https://github.com/jrl-umi3218/Tasks/pull/64

Note that this MR has an important implications for existing logs:
- The logged `CoM` position from the task (e.g `com_<RobotName>_pos`) is wrong along the dimensions where `dimWeight != 1`.
commit 643750d3 (2020-03-09): [CoMTask][bugfix] Fix logging/gui of CoM with dimWeight

Fixes a bug in logging/gui of the com position when using dimWeight.

Fixes #14
closed 2020-03-09 by gergondet