jrl-umi3218/mc_rtc #26: Bug in the mc_rtc with robot initial position

Saeed-Mansouri, 2020-04-13:
Hi @arntanguy @gergondet @stephane-caron 

I have encountered a problem in the mc_rtc package. I have imported my own robot into mc_rtc and in the robot.cpp file in mc_rtc/src/mc_robots folder and I am changing default joint angles of the robot.
These are two options that I have tested and my problem is the position of the robot in the world changes based on the mc_log_ui graphs. I have attached two figures which are xmin and xmax of support in X direction based on time of one or two steps and they show this change.
 1-std::vector<double> default_q = { 0.,0.,    0.,    0.08, -0.08,   0.0,    0.,     0.,    0.,   0.08,    -0.08,   0.}; 
2-std::vector<double> default_q={-0.0007,   -0.0013,   -0.7859,    0.5822,   -0.3200,    0.0014, -0.0007,   -0.0013,   -0.7859,    0.5822,   -0.3200,    0.0014}; 

With the first option robot starts from [0,0,z] and with second option it starts from [0.5,0,z]. The amount of change in the joints is not that much to change the upper body position with half a meter. But the position of the whole body and foot support position changes and I don't get why changing default joint angles should change position of the whole robot in the world. Do you have any idea what might be the problem here?

Thank You

[image]
[image]

arntanguy, 2020-04-13:
There is a bit of confusion there:

- `default_q` is the joints configuration defined in the same order as `_ref_joint_order`. In the case of JVRC from which I'm guessing you're copying to create your robot, a default joint order is generated from the URDF file.
- `_default_attitude` is the initial world pose of the floating base, defined as quaternion-translation vector: `{qw, qx, qy, qz, tx, ty, tz}`

What you want to be changing here is not `default_q` (joint values), but rather `_default_attitude` (floating base pose). Here what you're changing in your example is the angle of the 4th joint of your robot to `0.5822 rad`. Note that in `mc_log_ui`, you can look at the `ff` entry to see the position of the floating base.

Also, if you are creating a robot module for your own robot, you probably don't want to be doing that within mc_rtc itself, but rather as an independent repository. I think we could push the robot module for NAO robot to the public repository as an example.

Saeed-Mansouri, 2020-04-15:
It works. Thanks.

closed 2020-04-15 by Saeed-Mansouri