jrl-umi3218/mc_rtc #27: filtering FT data

Saeed-Mansouri, 2020-04-15:
Hi @arntanguy @gergondet @stephane-caron

I have found that the FT sensor data are noisy and affected the performance of the stabilizer. I should filter the FT data. I want to know that is any filter used in the mc_rtc for this purpose?

arntanguy, 2020-04-16:
We currently do not filter the force sensor measurement themselves. However, the `AdmittanceTask` and its derivatives, including the `CoPTask` used by the stabilizer, a very simple low-pass filter is applied on the reference velocity computed from the force sensor measurements (see `AdmittanceTask::update()` implementation.

```cpp
refVelB_ = 0.8 * refVelB_ + 0.2 * sva::MotionVecd(angularVel, linearVel)
```
where `linearVel` and `angularVel` are proportional the the wrench measurement error. This filter is currently hardcoded. This should not be the case, we simply haven't gotten around to making it configurable.

In addition, if your force sensor measurement are quite noisy, you may wish to filter those as well. In order for the stabilizer to use these filtered measurements, you'll need to filter the values of each `ForceSensor` instance in `robot().forceSensors()`. There are some very basic filters in the `mc_filter`, notably a `LowPass` filter which we currently use for filtering of the estimated CoM velocity in the `KinematicInertial` observer. It could be used to filter the force-sensor measurements as well if such a simple filter is appropriate for your measurements.

closed 2020-04-16 by Saeed-Mansouri

stephane-caron, 2020-04-16:
> However, the AdmittanceTask and its derivatives, including the CoPTask used by the stabilizer,

This part is actually trickier:

- ``AdmittanceTask`` corresponds to the hand wrench control from [Bouyarmane et al. (2018)](https://doi.org/10.1109/TRO.2018.2876782)
- ``DampingTask`` inherits from it, but redefines ``update()``, scraping the low-pass filter in the process (It is a discontinuity in the framework to pay attention to if refactoring :wink:). This task is closer to one described in Section II.D of [Kajita et al. (2010)](https://www.researchgate.net/profile/Kensuke_Harada/publication/224200078_Biped_walking_stabilization_based_on_linear_inverted_pendulum_tracking/links/53fe1bab0cf283c3583b709e.pdf)
- ``CoPTask`` inherits from ``DampingTask`` (no filter), adding the conversion described [here](https://hal.archives-ouvertes.fr/hal-01875387/document) (Section III.A)

The motivation for not having force filtering in the stabilizer is that low-pass filtering adds delay, which reduces the reactivity of balance control. In some cases it's unavoidable, see for instance the discussion in [Englsberger et al. (2018)](https://www.researchgate.net/profile/George_Mesesan/publication/325945431_Torque-Based_Dynamic_Walking_-_A_Long_Way_from_Simulation_to_Experiment/links/5dd53348a6fdcc37897c12ee/Torque-Based-Dynamic-Walking-A-Long-Way-from-Simulation-to-Experiment.pdf), but on HRPs we could do without it.