jrl-umi3218/mc_rtc #28: Suggestion for headless installation of mc_rtc

xeroxx, 2020-04-16:
Hi @arntanguy and @gergondet 
We're using mc_rtc framework with LIPM walking controller which is built from latest source code on their master branch. Recently we added our own robot model into the LIPM and It's working fine. We don't have any problems running it on a PC where we have the luxury of working with a graphical user interface and the mc_rtc panel RVIZ plugin. However we are trying to run the controller on a headless linux machine and we are not sure what is the best/effitient way of implementing a command line interface (cli) for the controller. Do you have any suggestion for this? Does mc_rtc framework have a cli interface to work with different controllers?

arntanguy, 2020-04-16:
The GUI/panel does not have to run on the same computer as the controller itself. If you have the possibility to communicate with the linux machine on which you are installing the controller, then you can launch on another machine:

```
roslaunch mc_rtc_ticker control_display.launch
```

The right click on the GUI, select "TCP" and enter the hostname of the controller's machine, and you'll have the GUI there. For the robot display you'll need to set-up `ROS_MASTER_URI/ROS_IP` appropriately.

Now if you do not have network access with the controller's machine, things would be a bit trickier. We don't really have a CLI interface to interact with the controller, and especially not the FSM part. For LIPMWalking, it is possible to make the controller start and execute a given plan automatically (we do this in continuous integration), but I'm guessing that wouldn't be enough for you.

gergondet, 2020-04-17:
Hi,

As Arnaud stated you can use the GUI remotely if you have network access from there (even if your embedded controller does not run with ROS functionalities enabled you can still access the GUI remotely, the only missing thing will be the robot configuration)

If that's not an option it's possible to use the same functionalities to build a simple CLI. I've made a super simple example [here](https://github.com/gergondet/lipm_walking_cli) to show how this can be done. This only enables start standing/walking, pause walking and plan selection but the principle can be applied to implement all the GUI functions.

xeroxx, 2020-04-17:
Thanks for your answer. With this, some tests can be run on the robot with another machine running the GUI.

xeroxx, 2020-04-17:
> The GUI/panel does not have to run on the same computer as the controller itself. If you have the possibility to communicate with the linux machine on which you are installing the controller, then [...]

Thanks for your answer. With this, some tests can be run on the robot with another machine running the GUI.

xeroxx, 2020-04-17:
> Hi,
>
> As Arnaud stated you can use the GUI remotely if you have network access from there (even if your embedded controller does not run with ROS functionalities enabled you can still access the GUI [...]

Thanks @gergondet ! This helped a lot.

closed 2020-04-17 by xeroxx