jrl-umi3218/mc_rtc #30: Minimum hardware requirements

xeroxx, 2020-04-18:
I couldn't find any information about minimum hardware requirements for a Linux machine to be able to provide a control loop frequency of 100-200 Hz. I currently have a BeagleBone Black SBC with mc_rtc + LIPM controller but it seems that the processor is now at full load. Do you guys have any suggestions for the hardware platform requirements?

gergondet, 2020-04-20:
Hi @xeroxx 

This is a good question but it is hard to answer. One major factor is how many degress of freedom the robot has. This will hugely impact the performance when solving the underlying optimization problem solved by mc_rtc. In addition, the LIPM controller runs a separate optimization for wrench distribution and one more for the MPC but these two are not affected by the numbers of dof in the robot.

Now, some numbers:
- on a regular desktop or laptop CPU (i.e. middle of the line Intel i5/i7) the LIPM controller runs at 500Hz with the HRP models (45~50 dofs)
- when running on HRP4, the controller does not run on the embedded CPU (Intel Pentium M 1.6GHz [[1]]) note that this is a really old CPU (I'm not sure about the exact model in HRP4 but that would have been released around 15 years ago), single-core without super-scaling and I think the BeagleBone Black could out-perform it. Furthermore, we would have to run at 200Hz on HRP4 and that would not be feasible but 100Hz might be although the single-core might be an issue

On the note of single core, the issue is that mc_rtc spawns typically three threads separate from the run function:
- logging thread (configured by `LogPolicy: threaded` option which is highly recommended when running real-time as you do not want to wait for writes to disk in your control loop)
- GUI thread
- ROS publication thread

Since the BeagleBone Black has a superscalar CPU it should be able to handle those without hindering the performance of the control loop. However, the thread where `controller.run()` is happening (i.e. your control loop) should have much higher priority.

You can modify the GUI refresh rate (`Timestep` parameter in the `GUIServer` section) and ROS publication rate (`timestep` parameter in the `publish` section of the ROS plugin parameters) to try and improve performance (or CPU load)

> it seems that the processor is now at full load.

You can look into the `perf_*` entries in the log to check how things are going or use `mc_bin_perf` to check them at a glance. I also strongly recommend that you compile with maximum optimizations enabled and something akin to `-march=native`

[1]: https://staff.aist.go.jp/k.kaneko/publications/2011_publications/IROS2011-0164.pdf

xeroxx, 2020-04-21:
Thanks, @gergondet.
I need some time to work on this. I will share the results with you.

gergondet, 2020-11-18:
Hi @xeroxx 

I will be closing this issue. If you have some results to share please feel free to do so.

Thanks

closed 2020-11-18 by gergondet