jrl-umi3218/mc_rtc #31: reading joint torque in mc_rtc

mohamadmahdavian, 2020-04-23:
Hi @arntanguy @gergondet 

I am trying to read the joint torques using mc_rtc functions. I am using this function:
controlRobot().mbc().jointTorques_[1][0]

But all I get is zero during robot simulation in choreonoid. Do you know what might be the reason here?

Thank You

arntanguy, 2020-04-23:
This is because the RTM output port for torques `tauOut` is not connected to the input port  `taucIn` of `mc_openrtm`  in your choreonoid project. On HRP robots (except HRP5), we do not have torque sensors, and their estimate from current isn't reliable, so we typically tend to ignore the torque inputs in our case ;)

I had a quick look at the JVRC project, and this was indeed the case here. See MR https://github.com/jrl-umi3218/mc_openrtm/pull/3 to see how it can be connected.

arntanguy, 2020-04-23:
Also note that if this works as expected, you should have an entry `tauIn` in the log file where you can see the torque for all joints.

mohamadmahdavian, 2020-04-23:
Thank You. The problem is solved

mohamadmahdavian, 2020-04-23:
By the way still I am getting zero from the controlRobot().mbc().jointTorque variable. Do you what might be the reason?
I think that tauIn data cones from choreonoid and in case we don't use choreonoid we may need to use that variable to calculate the torque.

arntanguy, 2020-04-23:
There are two quantities here:
- `robot().mbc().jointTorque` is the desired torque computed by the QP solver. Note that it is only computed if there is a dynamic constraint added to the QPSolver (`solver().addConstraintSet(dynamicsConstraint)`). Without dynamics constraints, this torque is always zero.
- `robot().jointTorques()` are the torque measured from sensor/choreonoid.

closed 2020-04-24 by mohamadmahdavian

mmahdavian, 2020-07-23:
Hi @arntanguy @gergondet 

I am encountering an issue with torque data in mc_log_ui. whenever I click on it in the gui in shows the torque data and when I click again it should to clear the graph. But it doesn't happen. Also, we wrote a code to extract all the data from the recorded files into csv files. For all of the variables we are able to do it. But for the torque data it has an error. I think there might be something minor wrong with the torque data in the GUI. Is it possible for you to check it out?

Thank You

[image]

gergondet, 2020-07-23:
Hi @mmahdavian 

> Also, we wrote a code to extract all the data from the recorded files into csv files. For all of the variables we are able to do it. But for the torque data it has an error. I

`mc_bin_to_log` does conversion from the `.bin` format to the `.csv` format. You can also look [here](https://jrl-umi3218.github.io/mc_rtc/tutorials/tools/mc_log_utils.html) for other ways to work with the log.

The reason you cannot extract the torque key is that `mc_log_ui` actually does a small trick to display those entries with friendly names based on the currently selected robot and the actual keys are named `tauIn_0`, `tauIn_1, ..., `tauIn_N`

> I am encountering an issue with torque data in mc_log_ui. whenever I click on it in the gui in shows the torque data and when I click again it should to clear the graph. But it doesn't happen.

This should have been fixed in 7c4baba379fe1fbb6b3bc6b457277069bcad0d3c