jrl-umi3218/mc_rtc #33: The components inside the mc_rtc panel in ROS RViz flashed and does not work.

kx-guan, 2020-04-25:
Hi, Dear everyone:
Thank you very much for discussion.

I compiled the new version (1.6) of LIPM stabilizer (https://github.com/stephane-caron/lipm_walking_controller), with new version (v1.2.1) of mc_rtc
(but replace GUIState.h by <mc_rtc/gui.h> as mentioned in LIPM stabilizer #35) using Ubuntu 16.04LTS.

When I tried to run simulation in V-REP, the components inside the mc_rtc panel in ROS RViz flashed (appear and disappear about once per second).
The marker in left side robot flashed with the components inside the mc_rtc panel, too. The robot itself did not flash. The title "mc_rtc panel" did not flash.
I tried to push the buttons in the mc_rtc panel, but the buttons did not work.
Last year, the previous version of LIPM stabilizer with the previous version of mc_rtc did not have such phenomenon (using Ubuntu 14.04LTS).
Could you give me any advice?
Thank you very much.
      Best regards.

gergondet, 2020-04-27:
Hi @kx-guan 

It could be that the simulation is running really slowly, can you check the output of `mc_bin_perf /tmp/mc-control-LIPMWalking-latest.bin` the next time you try to run the controller?

kx-guan, 2020-04-27:
Hi, Dear Dr. @gergondet. 
Thank you very much.
I gave two results of " mc_bin_perf /tmp/mc-control-LIPMWalking-latest.bin" as follows:

-----------------------------------------------------------------
|                     |  Average |    StdEv |      Min |    Max |
-----------------------------------------------------------------
|       ControllerRun |     94.5 |     75.2 |     44.4 |    586 |
|       FrameworkCost |     6.52 |     2.44 |     1.67 |   14.6 |
|           GlobalRun |     99.9 |     77.3 |     48.7 |    615 |
|                 Gui |   0.0236 |   0.0716 |  0.00226 |  0.729 |
|                 Log |     5.27 |      3.2 |     3.11 |   49.3 |
|    MPCBuildAndSolve |        0 |        0 |      inf |      0 |
|            MPCSolve |        0 |        0 |      inf |      0 |
|         Plugins_ROS |    0.136 |    0.138 | 0.000755 |  0.814 |
|   Plugins_ROS_after |    0.136 |    0.138 |  0.00781 |  0.813 |
|  Plugins_ROS_before | 0.000958 | 0.000515 |  0.00065 | 0.0114 |
| SolverBuildAndSolve |     52.6 |     59.5 |     15.7 |    419 |
|         SolverSolve |    0.362 |    0.265 |    0.193 |   7.06 |
|          Stabilizer |     1.39 |     2.96 |    0.645 |   47.5 |
-----------------------------------------------------------------

---------------------------------------------------------------------------------------
--------------------------------------------------------------
|                     | Average |   StdEv |      Min |   Max |
--------------------------------------------------------------
|       ControllerRun |     127 |    73.9 |     44.4 |   586 |
|       FrameworkCost |    5.11 |    2.31 |     1.56 |  14.6 |
|           GlobalRun |     133 |    75.9 |     48.7 |   615 |
|                 Gui |  0.0296 |  0.0872 |  0.00226 |  0.73 |
|                 Log |    5.64 |    3.04 |     3.11 |  49.3 |
|    MPCBuildAndSolve |       0 |       0 |      inf |     0 |
|            MPCSolve |       0 |       0 |      inf |     0 |
|         Plugins_ROS |   0.143 |   0.147 | 0.000755 |  1.23 |
|   Plugins_ROS_after |   0.142 |   0.147 |  0.00781 |  1.23 |
|  Plugins_ROS_before |   0.001 | 0.00121 | 0.000646 | 0.048 |
| SolverBuildAndSolve |    82.2 |    58.1 |     15.7 |   419 |
|         SolverSolve |     0.4 |   0.212 |    0.193 |  7.06 |
|          Stabilizer |    1.18 |    1.56 |    0.645 |  47.5 |
--------------------------------------------------------------

 Could you give me any advice?
 Thank you very much.
 Best regards.

gergondet, 2020-04-27:
Hi @kx-guan ,

These are pretty bad numbers, here's what I get on my machine for reference:
```
---------------------------------------------------------------
|                     |  Average |   StdEv |     Min |    Max |
---------------------------------------------------------------
|       ControllerRun |    0.328 |   0.266 |  0.0901 |   13.7 |
|       FrameworkCost |     21.8 |    8.06 |    1.24 |    100 |
|           GlobalRun |    0.426 |   0.457 |   0.123 |   22.4 |
|                 Gui |  0.00926 |  0.0221 | 0.00076 |  0.376 |
|                 Log |   0.0448 |   0.029 |  0.0126 |   1.01 |
|    MPCBuildAndSolve |    0.213 |  0.0723 |   0.142 |  0.382 |
|            MPCSolve |   0.0417 |  0.0145 |  0.0288 | 0.0906 |
|         Plugins_ROS |   0.0435 |   0.354 | 0.00229 |     22 |
|   Plugins_ROS_after |   0.0429 |   0.354 | 0.00212 |     22 |
|  Plugins_ROS_before | 0.000549 | 0.00303 | 7.5e-05 |  0.185 |
| SolverBuildAndSolve |    0.186 |  0.0847 |  0.0448 |   1.59 |
|         SolverSolve |   0.0383 |  0.0187 |  0.0162 |  0.727 |
|          Stabilizer |   0.0225 |  0.0151 | 0.00347 |  0.183 |
---------------------------------------------------------------
```

How did you compile mc_rtc and its dependencies? You should probably set `CMAKE_BUILD_TYPE` to `RelWithDebInfo`

kx-guan, 2020-04-27:
Hi, Dear Dr. @gergondet.
Thank you very much for quick response.
Yes, I compiled with the debug type. Now I recompiled mc_rtc and its dependencies with release type.Then I tested again. Now the panel does not flash again. And can see the robot walking.
This problem has been solved. 
Thank you very much again.
Best regards.

gergondet, 2020-04-28:
Hi @kx-guan 

That's great to hear. I'll close the issue then.

closed 2020-04-28 by gergondet