jrl-umi3218/mc_rtc #35: Changing TimeStep causes "QP failed to run()"

mohamadmahdavian, 2020-05-09:
Hello @arntanguy @gergondet @stephane-caron 

I am trying to change the timestep of the simulation of LIPM package running in choreonoid. I noticed that I need to change Timestep variable in mc_rtc.yaml file in mc_rtc/etc folder(Also I have changed the timestep or dt in all required files in the mc_openrtm and LIPM package) and I can see that the tiemstep has changed from 0.005 to 0.01 in the log_files in /tmp/mc-control-LIPMWalking-latest.bin file. But when I start the simulation, robot falls down and I get following error and I know it happens after the change in the mc_rtc.yaml file. Without that the timestep stays 0.005 sec and robot doesn't fall down even with all the changes in other files. Do you what might be the reason?

Thank You

```
Completed Initial
Starting state Standing
Starting Standing Init Function
Added task com_rh_r_v1
Added task cop_rh_r_v1_LeftFootCenter
Added task cop_rh_r_v1_RightFootCenter
Added task orientation_rh_r_v1_base_link
cop_rh_r_v1_LeftFootCenter: clamping y angular velocity to -0.2
cop_rh_r_v1_RightFootCenter: clamping y angular velocity to -0.2
cop_rh_r_v1_LeftFootCenter: y angular velocity back within range
cop_rh_r_v1_RightFootCenter: y angular velocity back within range
cop_rh_r_v1_LeftFootCenter: clamping y angular velocity to -0.2
cop_rh_r_v1_RightFootCenter: clamping y angular velocity to -0.2
cop_rh_r_v1_LeftFootCenter: y angular velocity back within range
cop_rh_r_v1_RightFootCenter: y angular velocity back within range
cop_rh_r_v1_LeftFootCenter: clamping y angular velocity to -0.2
cop_rh_r_v1_RightFootCenter: clamping x angular velocity to -0.2
cop_rh_r_v1_RightFootCenter: clamping y angular velocity to -0.2
cop_rh_r_v1_LeftFootCenter: clamping x angular velocity to 0.2
cop_rh_r_v1_LeftFootCenter: y angular velocity back within range
cop_rh_r_v1_RightFootCenter: x angular velocity back within range
cop_rh_r_v1_RightFootCenter: y angular velocity back within range
cop_rh_r_v1_LeftFootCenter: x angular velocity back within range
cop_rh_r_v1_LeftFootCenter: clamping y angular velocity to -0.2
cop_rh_r_v1_RightFootCenter: clamping y angular velocity to -0.2
cop_rh_r_v1_LeftFootCenter: y angular velocity back within range
cop_rh_r_v1_RightFootCenter: clamping x angular velocity to -0.2
cop_rh_r_v1_RightFootCenter: y angular velocity back within range
cop_rh_r_v1_LeftFootCenter: clamping x angular velocity to 0.2
cop_rh_r_v1_LeftFootCenter: clamping y angular velocity to -0.2
cop_rh_r_v1_RightFootCenter: x angular velocity back within range
cop_rh_r_v1_RightFootCenter: clamping y angular velocity to -0.2
cop_rh_r_v1_RightFootCenter: y angular velocity back within range

QP failed to run()

```

arntanguy, 2020-05-09:
Changing the timestep in `mc_rtc.yaml` changes it for the QP, but not the choreonoid simulation. Here this mismatch is causing the foot force control to apparently diverge, which in turn leads to an infeasible QP (some constraints are violated).

You'll also want to change the timestep for the choreonoid simulation. I believe you need to modify the `PDcontroller.conf.choreonoid` file used by your simulation.

mohamadmahdavian, 2020-05-09:
I have changed all the timesteps in the cnoid folder of mc_openrtm and also the rates in RobotHardware.conf file and sim_mc.cnoid files to 100hz but I still see this issue.

mohamadmahdavian, 2020-05-09:
@stephane-caron I have changed the timestep values for choreonoid files. Do you know what might be the reason of problem here? I have attached the simulation files for you here.
Thank You
[simulation_files.zip](https://github.com/jrl-umi3218/mc_rtc/files/4602827/simulation_files.zip)

stephane-caron, 2020-05-15:
Your simulation files look correct to me. You can check that the `periodicRate` of your robot is correctly set in your Choreonoid simulation via the `Property` tab:

[image]

On the mc_rtc side, make sure you print out your ``controller().timeStep`` and that it has your desired value of 0.01 [s].

To make sure that all tasks are working properly, I would also add some tasks from the mc_rtc GUI while the controller is in `Initial` state. For instance, add two SurfaceTransformTasks for the pelvis and a hand, and move the hand target around in RViz. Task gains should be independent from the controller time step, but there's never harm in checking that everything works properly :wink:

mohamadmahdavian, 2020-05-15:
@stephane-caron @arntanguy  Thank you for your reply. What solved the problem was adjusting task values in LIPMWalking.conf file. Now the robot is walking with new time step. Thank you for your advises.

closed 2020-05-15 by mohamadmahdavian