jrl-umi3218/mc_rtc #38: Running tests and getting data output

kurt, 2020-05-27:
I am wondering how to get the results from the mc_rtc/tests. How are the test results from the boost libraries recorded and where could I find or generate appropriate xml files 

Thanks

gergondet, 2020-05-28:
Hi @kurt 

There's no such things being recorded, Boost is only used to run the test code and check the expected results. Could you explain why you would need such files?

kurt, 2020-05-28:
Hi I am just looking into making sure the code is safe when relied upon for safety critical applications. thus interested in unit tests, code coverage etc. I have figured a way of doing it which could be added to the repo if more people are interested but it will take a bit of time to finalize. Its a matter of just exporting the test results to an appropriate xml file and using tools like gcov and lcov to ensure the whole system has been tested. Thanks

gergondet, 2020-05-29:
In that case you can add `--coverage` flags to the compilation (i.e. manipulate `CMAKE_CXX_FLAGS`, `CMAKE_EXE_LINKER_FLAGS` and `CMAKE_MODULE_LINKER_FLAGS`) and that should generate a `coverage.info` which can be fed to lcov to generate a report. ATM we unfortunately don't have coverage reports enabled/uploaded in CI but that would be a welcome addition.

Note: this is assuming you can work with GCC or clang, I don't know if there is an equivalent MSVC flag.

kurt, 2020-05-29:
Sounds good! I will add it to the database if I get it running properly on my end. Thanks!

kurt, 2020-06-06:
I am wondering if you or someone else could explain to me how the folder linking/header file linking all works for the CMake build so that I can add gcov properly?

kurt, 2020-06-07:
This is how I did it (although need to double check it did all I wanted it to) - in the mc_rtc/build folder : 
`cmake - GENERATE_COVERAGE=ON ..`
`make`
`ctest` 
`lcov --directory .. --capture --output-file testout`
`genhtml testout --output-directory html`
Then open the index.html file in the html folder and all the src/ files will show the code coverage of the LIPM code

closed 2020-06-07 by kurt