jrl-umi3218/mc_rtc #42: Question about _default_attitude

elfalko, 2020-06-29:
Hi, I'm currently trying to integrate a ReemC robot into the mc_rtc framework (with ros support).

I got the example of turning the head working. For the CoM Task however I need to shift the robot upwards, so it is not halfway stuck in the ground plane.

In the tutorial, this seems to be accomplished by the line 
```
_default_attitude = {{1., 0., 0., 0., 0., 0., 0.8275}};
```
in the robot module.
However, that does nothing for me. Changing values (assuming this is a {<quaternion>,<offset-vector>} notation) does not result in any change of the robot's position.

I managed to move the robot to a more realistic position with
```
robots().robot(0).posW(sva::PTransformd(sva::RotZ(0), Eigen::Vector3d(0., 0., 0.85))); 
```
in the controller, so the values seem about right. But the moment I actually define the contact surfaces in the controller as well, the entire thing fails with QP crashing.
I also tried different Z offsets in case the robots feet didn't reach the ground or something like that, but it didn't work either.
If I do not define contact surfaces, the robot just hangs in the air, limbs moving but with a fixed COM. Maybe the posW command nails it in place.
What am I missing here?

### source files

<details>
  <summary>reemc.h</summary>

``` c++
  #pragma once

  #include <mc_rbdyn/RobotModule.h>
  #include <mc_rbdyn/RobotModuleMacros.h>
  #include <mc_rtc/logging.h>
  #include <mc_rbdyn_urdf/urdf.h>
  #include <mc_robots/api.h>

  namespace mc_robots{
    struct MC_ROBOTS_DLLAPI ReemcModule : public mc_rbdyn::RobotModule{
      public:
        ReemcModule();

        mc_rbdyn_urdf::Limits limits;
      protected:
        void readUrdf(const std::string & robotName, bool fixed, const std::vector<std::string> & filteredLinks);

    };
  }

  extern "C"{
    ROBOT_MODULE_API void MC_RTC_ROBOT_MODULE(std::vector<std::string> & names){
      // Names of the robots exported by this module
      names = {"Reemc"};
    }

    ROBOT_MODULE_API void destroy(mc_rbdyn::RobotModule * ptr){
      delete ptr;
    }

    ROBOT_MODULE_API mc_rbdyn::RobotModule * create(const std::string & name){
      ROBOT_MODULE_CHECK_VERSION("Reemc")

      // At this point name must be one of the supported robots
      if(name == "Reemc"){
        return new mc_robots::ReemcModule();
      }else{
        LOG_ERROR("Reemc module cannot create object of type " << name)
        return nullptr;
      }
    }
  }
```
</details>

<details>
  <summary>reemc.cpp</summary>

``` c++

  #include "reemc.h"

  #include "config.h"

  #include <RBDyn/parsers/urdf.h>

  #include <fstream>

namespace mc_robots{
  ReemcModule::ReemcModule() : mc_rbdyn::RobotModule(REEMC_DESCRIPTION_PATH, "reemc"){
      init(rbd::parsers::from_urdf_file(REEMC_URDF_PATH));

  std::vector<double> default_q = {
    0., 0., 0., 0., 0., 0., // right leg (from foot to hip)
    0., 0., 0., 0., 0., 0., // left leg (from foot to hip)
    0., 0., //torso
    0., 0., //head
    0.5*pi, 0.4*pi, 0., 0., 0., 0., 0., // right arm (from shoulder to wrist)
    0.5*pi, 0.4*pi, 0., 0., 0., 0., 0. // left arm (from shoulder to wrist)
  };

  const auto & rjo = ref_joint_order();

  for(size_t i = 0; i < rjo.size(); ++i){
    _stance[rjo[i]] = {default_q[i]};
  }

  //prob orientation(w,x,y,z), offset
  _default_attitude = {{1., 0., 0., 0., 0., 0., 0.85}};

  auto da=default_attitude();
  LOG_SUCCESS("default attitude x:"<< da[4]<<" y:"<<da[5]<<"z:"<<da[6]);

  _forceSensors.push_back(mc_rbdyn::ForceSensor("RightFootForceSensor", "right_sole_link", sva::PTransformd::Identity()));
  _forceSensors.push_back(mc_rbdyn::ForceSensor("LeftFootForceSensor", "left_sole_link", sva::PTransformd::Identity()));

  _bodySensors.emplace_back("Accelerometer", "imu_link", sva::PTransformd(Eigen::Vector3d(-0.0325, 0, 0.1095)));
  _bodySensors.emplace_back("FloatingBase", "base_link", sva::PTransformd::Identity());

  _minimalSelfCollisions = {
  }; 

  _commonSelfCollisions = _minimalSelfCollisions;

  /* // Configure the stabilizer. Uses the default values of the */
  /* // StabilizerConfiguration except for where values specific to the JVRC */
  /* // robot are required. */
  _lipmStabilizerConfig.leftFootSurface = "LeftFootCenter";
  _lipmStabilizerConfig.rightFootSurface = "RightFootCenter";
  _lipmStabilizerConfig.torsoBodyName = "base_link";
  _lipmStabilizerConfig.comHeight = 0.85;
  _lipmStabilizerConfig.comActiveJoints = {
    "leg_right_1_joint",
    "leg_right_2_joint",
    "leg_right_3_joint",
    "leg_right_4_joint",
    "leg_right_5_joint",
    "leg_right_6_joint",
    "leg_right_7_joint",
    "leg_left_1_joint",
    "leg_left_2_joint",
    "leg_left_3_joint",
    "leg_left_4_joint",
    "leg_left_5_joint",
    "leg_left_6_joint",
    "leg_left_7_joint",
    "torso_1_joint",
    "torso_2_joint"
  };
  _lipmStabilizerConfig.torsoPitch = 0;
  _lipmStabilizerConfig.copAdmittance = Eigen::Vector2d{0.01, 0.01};
  _lipmStabilizerConfig.dcmPropGain = 5.0;
  _lipmStabilizerConfig.dcmIntegralGain = 10;
  _lipmStabilizerConfig.dcmDerivGain = 0.5;
  _lipmStabilizerConfig.dcmDerivatorTimeConstant = 1;
  _lipmStabilizerConfig.dcmIntegratorTimeConstant = 10;
}
}
```

<details>

gergondet, 2020-06-29:
Hi @elfalko 

The following line loads the robot with a fixed base (default value of the [fixed argument](https://github.com/jrl-umi3218/RBDyn/blob/master/src/parsers/RBDyn/parsers/urdf.h#L62)):

```cpp
init(rbd::parsers::from_urdf_file(REEMC_URDF_PATH));
```

In order to get it to work you should call init with the fixed argument `false`:

```cpp
init(rbd::parsers::from_urdf_file(REEMC_URDF_PATH, false));
```

> But the moment I actually define the contact surfaces in the controller as well, the entire thing fails with QP crashing

This is a little surprising, let us know if you still encounter the issue after fixing the loading issue. In case you still have this issue, can you share the model and surfaces definition that you are using?

elfalko, 2020-07-01:
Thanks, that was the part I was missing!
I knew you could initialize a fixed robot as the example RobotModule code has the fixed parameter, but it never occurred to me to check the `from_urdf_file` definition as my urdf was loaded and seemed ok to me.
After I changed the parameter to false, my robot bent over strangely backwards, like Neo from the Matrix dodging bullets, and QP still kept crashing.
[image] 
But after I set some more sensible joint angle defaults (basically legs bent a little, not totally stretched) it worked just fine.
[image]

closed 2020-07-01 by elfalko