jrl-umi3218/mc_rtc #43: Connecting mc_rtc to a custom robot

magnusmusngi, 2020-07-03:
Hi,

We are planning to use mc_rtc in our robot that uses a custom made interface board and is connected to a PC through an ethernet port. What is the best way to achieve this connection, and is there available documentation/guide and files that could help in implementing this?

Also, when we implement this, we anticipate that there will be an issue with jitter when it comes to processing data in the PC and sending it to the robot's interface board. Do you have suggestions on how to setup a RT-patched Linux to deal with this issue? We already tried running mc_rtc on a RT_PREEMPT kernel with RT priority before, but it did not seem to work.

Thanks

arntanguy, 2020-07-03:
Unfortunately, we haven't written the documentation on how to create your own custom interface yet. Fortunately, it is pretty easy to do. The role of this interface is to read sensor values from the robot, and provide commands from `mc_rtc`. This is done through the `MCGlobalController` class.

1- Create and `MCGlobalController controller` instance
2- Read the robot encoder state and provide is to the global controller (`MCGlobalController::init`)
3- Start a control loop, this should:
    a. Read sensors values (encoders, force sensors, etc), provide them to the global controller
    b. Run the controller (eg call `MCGlobalController::run`). This will delegate control to the active `MCController` instance as defined in your `mc_rtc.yaml` configuration file
    c. Provide the controller results from `controller.robot().mbc()` (joints `q`, torques `tau`, etc)

You may refer to [mc_franka](https://github.com/gergondet/mc_franka/blob/master/src/MCFrankaControl.cpp) for a simplistic interface, or the more complete [mc_udp](https://github.com/jrl-umi3218/mc_udp/blob/master/src/client/MCUDPControl.cpp) or [mc_openrtm](https://github.com/jrl-umi3218/mc_openrtm)

Regarding the real-time issue, what you'll want is to have the control-loop of step `3` above running in a high-priority thread. Of the interfaces mentioned above, we only do it with `mc_openrtm` that we use to run the controller embdeded on our robot's computer. There the [onExecute()](https://github.com/jrl-umi3218/mc_openrtm/blob/master/src/MCControl.cpp#L166) functions runs with such priority. I am however not very familiar with how to set-up such a real-time priority manually.

Regarding jitter, you'll want to make sure that your controller (`MCControl::run`) computes much faster than the control loop frequency of your robot. Typically for running a controller deported with our HRP-2 robot (using `mc_udp`) controlled at `200Hz (5ms)` we have
- Controller `~2ms` computation time
- Robot control loop `5ms`
This leaves enough time for communication to read the sensor values and provide the new commands, and avoids missing iterations (at least over ethernet communication)

gergondet, 2020-07-03:
> Do you have suggestions on how to setup a RT-patched Linux to deal with this issue? We already tried running mc_rtc on a RT_PREEMPT kernel with RT priority before, but it did not seem to work.

Could you elaborate on this?

I have published a simple sample for Linux with PREEMPT RT that I have used successfully in the past [here](https://github.com/gergondet/mc_linuxrt)

If you are running in real-time, it's also important to set the `LogPolicy` to `threaded` to avoid computation spikes due to disk flushing

magnusmusngi, 2020-09-16:
Thanks @arntanguy.

> Unfortunately, we haven't written the documentation on how to create your own custom interface yet. Fortunately, it is pretty easy to do. The role of this interface is to read sensor values from the robot, and provide commands from mc_rtc. This is done through the MCGlobalController class.

> You may refer to mc_franka for a simplistic interface, or the more complete mc_udp or mc_openrtm

Does this also provide a similar functionality like how RViz (the one executed through the command `rosrun mc_rtc_ticker control_display.launch`) could send commands through the buttons in the GUI, which then results to a callback inside a state? Because I would like to achieve something similar to what the buttons in RViz does (and having a corresponding callback inside the state), except that I would use a different UI.

Thank you.

gergondet, 2020-09-17:
Hi @magnusmusngi 

This functionality is independent of mc_rtc_ticker and is part of mc_rtc.

In fact the RViZ panel is available as a stand-alone Qt application:
- with catkin/ROS on your system you have it already `rosrun mc_rtc_rviz_panel mc_rtc_gui`, this version still publishes ROS markers/interactive markers
- you can build a fully ROS free version

All of this is based on Qt. If you want to implement a different UI you can inherit from [`mc_control::ControllierClient`](https://github.com/jrl-umi3218/mc_rtc/blob/master/include/mc_control/ControllerClient.h) which you can also use to build a very specialized UI for a specific controller.

magnusmusngi, 2020-09-22:
Thanks, @gergondet.

> All of this is based on Qt. If you want to implement a different UI you can inherit from mc_control::ControllierClient which you can also use to build a very specialized UI for a specific controller.

To do this, do I also have to deal with `mc_control::ControllerServer`? Also, what do I use as the values for `category`, `name`, and `sid` fields of `mc_control::ElementId` when I call `ControllerClient::send_request`? Additionally, could I still retain my calls to `StateBuilder::addElement` in my controller code if I want to have the same elements in my own UI?

Thank you.

gergondet, 2020-09-22:
> To do this, do I also have to deal with `mc_control::ControllerServer`?

No. `mc_control::ControllerServer` is on the controller side. You are only writing a new client.

> Also, what do I use as the values for category, name, and sid fields of mc_control::ElementId when I call ControllerClient::send_request?

The simplest way is to use the helper in `mc_control::ControllerClient` to send requests, e.g. [`ButtonWidget::button_released`](https://github.com/jrl-umi3218/mc_rtc_ros/blob/master/mc_rtc_rviz_panel/src/ButtonWidget.cpp#L18), [`TransformInteractiveWidget::handleRequest`](https://github.com/jrl-umi3218/mc_rtc_ros/blob/master/mc_rtc_rviz_panel/src/TransformInteractiveMarkerWidget.cpp#L36), [`NumberSliderWidget::sliderMoved`](https://github.com/jrl-umi3218/mc_rtc_ros/blob/master/mc_rtc_rviz_panel/src/NumberSliderWidget.cpp#L42). The gist of it is you use the `mc_control::ElementId` to identify the element and sends data along if needed by the element.

> Additionally, could I still retain my calls to StateBuilder::addElement in my controller code if I want to have the same elements in my own UI?

Yes. This is transparent to the controller

magnusmusngi, 2020-09-22:
Thanks, @gergondet,

> The simplest way is to use the helper in mc_control::ControllerClient to send requests, e.g. ButtonWidget::button_released, TransformInteractiveWidget::handleRequest, [...]

My apologies. I wasn't able to mention that the UI that I will be using is a physical/hardware-based UI (at least with buttons), and not a completely software-based one. I would like to have in the physical UI a subset of the functionality that I have in the RViz GUI.

How I understood it so far is that a call to `client().send_request(id())` is what initiates the sending of a button press event from the GUI to the controller, and it is the `id()` (the `ElementId`) that maps the GUI button to the corresponding call to `StateBuilder::addElement` in the controller. Could achieving the same functionality in the hardware UI be done by just calling `ControllerClient::send_request()` and I just need the proper `mc_control::ElementId` (with the TCP connection already established)? For this case, what do I use for the field values of `ElementId`? I assume the categories will be just the once that I use during `AddElement` calls in the controller. How about for the `name` and `sid` fields.

Thank you.

gergondet, 2020-09-23:
@magnusmusngi 

Yes. You understood it correctly. Using `client().send_request(id)` with a connected client will be the same as clicking the button in the RViZ panel for this element.

> I assume the categories will be just the once that I use during AddElement calls in the controller. How about for the name and sid fields.

Yes. `name` is also the name of the element that you give. `sid` is irrelevant for sending requests and `mc_control::ElementId` has a constructor that doesn't use it.

Here's a simple example that removes the CoM task with the JVRC1 robot in the sample CoM controller:

```c++
#include <mc_control/ControllerClient.h>                                                                              
#include <thread>

struct SpecialClient : public mc_control::ControllerClient                                                            
{                                                                                                                     
  /** Same constructors as the base class */                                                                          
  using mc_control::ControllerClient::ControllerClient;                                                               

  /** We only override this and do nothing */
  void category(const std::vector<std::string> &, const std::string &) override {}                                    
};

int main()
{
  SpecialClient client("ipc:///tmp/mc_rtc_pub.ipc", "ipc:///tmp/mc_rtc_rep.ipc");
  // Connection is asynchronous so we wait a bit for it to roll                                   
  std::this_thread::sleep_for(std::chrono::seconds(1));                                                               
  client.send_request({{"Tasks", "com_jvrc1"}, "Remove from solver"});                                                
  // Similarly, send_request is asynchronous
  std::this_thread::sleep_for(std::chrono::seconds(1));                                                               
  return 0;
}
```

gergondet, 2020-11-18:
Hi @magnusmusngi 

I will be closing this issue. Feel free to re-open it if you have any further questions.

Note that since #86 you can also:
- use a synchronous [`run`](https://github.com/jrl-umi3218/mc_rtc/blob/master/include/mc_control/ControllerClient.h#L110) method to check for new message from the server (and it's also much easier to integrate in your own "main loop")
- override [`default_impl`](https://github.com/jrl-umi3218/mc_rtc/blob/master/include/mc_control/ControllerClient.h#L629) to silence warnings about unimplemented elements

closed 2020-11-18 by gergondet