jrl-umi3218/mc_rtc #49: Issue with tutorial - ROS not connecting

kurt, 2020-07-21:
I am trying to finish the "[Your first controller with mc_rtc](https://jrl-umi3218.github.io/mc_rtc/tutorials/introduction/first-controller.html )" I did the steps but only get this:
[image]

kurt, 2020-07-21:
Cannot convert from YAML if the root type is not a map or a sequence
Configuration dump until the attempted conversion:
{} was an issue with my YAML having "---" at the top of it.

Now I am just getting: 
[image]

kurt, 2020-07-21:
The issue is here in mc_rtc_ticker.cpp : `qOut[i] = controller.robot().mbc().q[i+1][0];`

arntanguy, 2020-07-22:
Hi @kurt,

I'm not sure where you got your `mc_rtc_ticker` code from, but if you look on the master branch this is not the current implementation of [mc_rtc_ticker.cpp](https://github.com/jrl-umi3218/mc_rtc_ros/blob/master/mc_rtc_ticker/src/mc_rtc_ticker.cpp).
And I don't recall it ever having had a `qOut` vector: there would be no point for it since it does not need to send any joint commands to a robot, it merely runs the `MCController::run()` function, which in turns runs the `QPSolver::run()` function that solves the QP problem, and performs double integration of its desired joint accelerations to obtain the new position of all robot joints (purely open-loop control scheme).

Are you sure you don't have any custom local changes? Can you try with the official master branch, there shouldn't be any problems there.

closed 2020-07-28 by arntanguy

kurt, 2020-07-30:
the issue was in the control code here... 
 `bool MyFirstController::run()
{
  if(std::abs(postureTask->posture()[jointIndex][0] - robot().mbc().q[jointIndex][0]) < 0.05)
  {
    switch_target();
  }
  return mc_control::MCController::run();
}`
the tutorial doesnt seem to work anymore.... I tried to follow it but was unsuccessful. Perhaps this issue is resolved with more recent code as you mention. I moved on from this and started calling the controllers using a similar method found in the tests/controllers (Cmakelists). Thanks