jrl-umi3218/mc_rtc #55: RBDyn must be built with parsers support

kurt, 2020-08-05:
This issue I am getting while trying to build mc_rtc. I am not sure why but I ran the build_and_install script and it wont build anymore. The build_and_install has an issue as well...

 I am getting another issue from the build and install which is: 
CMake Error at cmake/cython/cython.cmake:88 (add_custom_target):
  Cannot find source file:

    eigen_qld/__init__.py

arntanguy, 2020-08-05:
We recently deprecated `mc_rbdyn_urdf` package and integrated its parsers directly as part of `RBDyn`.
It's likely that you have some previously installed cmake files that weren't removed properly (this can happen if you've installed by hand without running `make uninstall` first)

You can try to remove the installed files first, then re-run the script:
```
# Removes installed cmake files 
sudo rm -rf  /usr/local/lib/cmake/RBDyn /usr/local/lib/cmake/Tasks /usr/local/lib/cmake/mc_rtc
# Removes installed libraries 
sudo rm -rf /usr/local/lib/libRBDyn* /usr/local/lib/libTasks* /usr/local/lib/libmc_*
```

kurt, 2020-08-05:
This caused some sort of break in my make file... should I rerun the build_and_install script? I did and it gave me:
```
CMake Error at cmake/cython/cython.cmake:88 (add_custom_target):
  Cannot find source file:

    eigen_qld/__init__.py

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx
Call Stack (most recent call first):
  cmake/cython/cython.cmake:283 (_ADD_CYTHON_BINDINGS_TARGETS)
  binding/python/CMakeLists.txt:5 (ADD_CYTHON_BINDINGS)

-- Generating done
-- Build files have been written to: /home/username/lipm/eigen-qld/build
-- [ERROR] CMake configuration failed for eigen-qld
Installation failed.
Installation log has been written to /tmp/build_and_install_warnings-2020-08-05-09-51-59.log
Please fix the issues and re-run the script.

```

kurt, 2020-08-05:
I am just getting this when I deleted the task folders
 

```
CMake Error at cmake/package-config.cmake:67 (find_package):
  By not providing "FindTasks.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Tasks", but
  CMake did not find one.

  Could not find a package configuration file provided by "Tasks" with any of
  the following names:

    TasksConfig.cmake
    tasks-config.cmake
```

kurt, 2020-08-05:
it seems as now I have broken the repository is there a set of steps to properly update the repository? Thanks :)

arntanguy, 2020-08-06:
Yes, you need to rerun the script, above we're removed the cmake scripts and libraries for mc_rtc and its main dependencies. Make sure that you're on the master branch of mc_rtc and up-to date before running the script.

Assuming you have no local changes:
```
cd src/mc_rtc
git pull origin master
git submodule sync
```

Then run the script.

kurt, 2020-08-07:
There still are issues.... are there commits that I could progress from where I am to have the build work again? My last pull was from March.

kurt, 2020-08-14:
I just started with a fresh install

closed 2020-08-14 by kurt