jrl-umi3218/mc_rtc #57: POSIX compliance of mc_rtc

magnusmusngi, 2020-08-07:
Hi,

We are planning to run mc_rtc into a non-Linux or non-Unix, but POSIX compliant OS. For this, I was wondering if mc_rtc is POSIX compliant, such that we could port it into this OS.

Thank you

gergondet, 2020-08-08:
Hi @magnusmusngi 

I think it should be ok if you can get all of the dependencies of mc_rtc in this environment and if it has a sufficiently recent compiler (mc_rtc is tested with GCC >= 4.8.0, clang >= 3.5.0 and Visual Studio >= 2019). Your best bet is to try it out.

magnusmusngi, 2020-08-13:
Thanks, @gergondet 

Were there any known instances of mc_rtc getting successfully ported to OSs like QNX or VxWorks?

Thanks.

gergondet, 2020-08-14:
> Were there any known instances of mc_rtc getting successfully ported to OSs like QNX or VxWorks?

Not that I'm aware of. From what I can gather, both VxWorks and QNX recent releases have modern toolchains and it seems Boost also supports both platforms so it should be mostly fine but it might also require some work.

gergondet, 2020-11-18:
Hi @magnusmusngi 

I will be closing this issue. Please open a new one if you encounter build issues on QNX and/or VxWorks and we will do our best to help you.

closed 2020-11-18 by gergondet