jrl-umi3218/mc_rtc #65: A tutorial on FSM controller

fkanehiro, 2020-09-15:
@arntanguy , as @mmurooka pointed out before, a tutorial on FSM controller which will appear https://jrl-umi3218.github.io/mc_rtc/tutorials/recipes/fsm-example.html is important to understand FSM in mc_rtc. Could you please write it at your earliest convenience?

closed 2020-09-19 by arntanguy

fkanehiro, 2020-09-20:
Thanks!
How the initial state in OpenDoorFSM is chosen?

arntanguy, 2020-09-21:
I forgot to mention it. There is the following tank property:

```yaml
init: FirstStateName
```

If it is missing then the first state in the `states` section would be used.

fkanehiro, 2020-09-21:
Door_Initial is not in the states section. How it is chosen?

arntanguy, 2020-09-21:
The available states are either loaded from libraries or declared as YAML.
- The ones installed by default with mc_rtc (e.g `MetaTasks`, `Parallel`, etc)
- The C++/python ones exported by your controller (e.g `Door_Inital` exported in `src/states/CMakeLists.txt`)
- The YAML states from the `states` section (or installed yaml files). Note that all yaml states are based on existing C++ states, and merely configure them.

The paths where `mc_rtc` looks for is defined in the controller's FSM configuration file. mc_rtc loads all libraries found in these path, and looks for the states loading symbols, defined by the macro

```
EXPORT_SINGLE_STATE("Door_Initial", Door_Initial)
```

This means that the states don't have to be part of `mc_rtc`, or a controller, they can be exported by any project (plugins, planners, etc.). For instance for the torquing demo, I have a plugin to manage the visual servoing, that receives the pose of markers over ROS, and provides states to perform the visual servoing. The controller then simply uses these states and configures them from the YAML configuration. We can discuss it in more details when I'm back next week.

PR #67 (2020-09-19): [doc] Add multi-robot door opening tutorial and sample controller
This PR:
- Adds the missing "FSM in practice" tutorial: https://jrl-umi3218.github.io/mc_rtc/tutorials/recipes/fsm-example.html
This shows how to exploit the FSM feature and multi-robot QP to open a door.
- Provide a new `DoorSample` controller that implements the tutorial's example.

Additional changes:
- Adds a `PostureState` to change the gains/targets of the global posture task
- Generate a fixed-based `env` variant of the `object` robots.
- `HalfsittingState` can now be used for robots other than the main robot.

Fixes #65