jrl-umi3218/mc_rtc #68: FSM visualization tool

fkanehiro, 2020-09-21:
Is there a tool for FSM visualization?

arntanguy, 2020-09-23:
We have a work in progress tool in development here: https://gite.lirmm.fr/atanguy/fsm-editor (I gave you access). So far it can display the states in the FSM and how they are interconnected, and allows to choose the depth level at which it is being displayed, and interactively expand specific states  (clicking on the `+` button). As-is it's a very crude tool, but it can be easily improved and extended.

For example, here is the [AdmittanceSample FSM](https://github.com/jrl-umi3218/mc_rtc/blob/master/src/mc_control/samples/Admittance/etc/AdmittanceSample.in.yaml)
[image]

And for a more complex example, here is the [LIPMStabilizer sample FSM](https://github.com/jrl-umi3218/mc_rtc/blob/master/src/mc_control/samples/LIPMStabilizer/etc/LIPMStabilizer.in.yaml)

[image]

[image]

fkanehiro, 2020-09-24:
Thank you @arntanguy , it looks very nice. But it seems to be incompatible with the latest mc_rtc.
I'll submit an issue to the repository later.

closed 2020-09-24 by fkanehiro

arntanguy, 2020-09-24:
Yes, for the time being, it requires a branch https://github.com/arntanguy/mc_rtc/tree/topic/FSMEditor. I've rebased it on top of the current master.