jrl-umi3218/mc_rtc #71: Build error installation

jari-van-steen, 2020-09-23:
I'm trying to install mc_rtc release 1.4.0 on Ubuntu 18.04.5 LTS. However, building from the script provided in the tutorial results in an error, where it says tests 20-39 fail. I included the log-file and copied the error message below:

[build_and_install_warnings-2020-09-23-11-39-07.log](https://github.com/jrl-umi3218/mc_rtc/files/5267507/build_and_install_warnings-2020-09-23-11-39-07.log)

The following tests FAILED:
	 20 - TestPostureController (SEGFAULT)
	 21 - TestPostureControllerLog (Failed)
	 22 - TestPostureControllerThreadedLogging (SEGFAULT)
	 23 - TestPostureControllerThreadedLog (Failed)
	 24 - TestPostureController2 (SEGFAULT)
	 25 - TestPostureControllerControllerSwitchThreadedLog (Failed)
	 26 - TestPostureController2ControllerSwitchThreadedLog (Failed)
	 27 - TestCoMTaskController (SEGFAULT)
	 28 - TestPositionTaskController (SEGFAULT)
	 29 - TestOrientationTaskController (SEGFAULT)
	 30 - TestEndEffectorTaskController (SEGFAULT)
	 31 - TestObserverController (SEGFAULT)
	 32 - TestSampleController_CoM (Failed)
	 33 - TestSampleController_Body6d (Failed)
	 34 - TestSampleController_Posture (Failed)
	 35 - TestSampleController_Text (Failed)
	 36 - TestSampleController_FSM (Failed)
	 37 - TestSampleController_LIPMStabilizer (Failed)
	 38 - TestPythonController (Failed)
	 39 - TestPythonLog (Failed)
Errors while running CTest
-- [ERROR] Testing is still failing for mc_rtc, please investigate deeper
Installation failed.
Installation log has been written to /tmp/build_and_install_warnings-2020-09-23-11-39-07.log
Please fix the issues and re-run the script.

I have no clue what the issue could be, so I would love to know what I could do to fix it. Thanks in advance!

gergondet, 2020-09-23:
Hi @jari-van-steen 

Thanks for attaching the build log, it's very useful to pin down the issue.

Looking at `TestPostureController`, we have:

```
test 20
      Start 20: TestPostureController

20: Test command: /home/jarivansteen/mc_rtc/build/tests/test_controller_ticker "--run_test=RUN" "--" "/home/jarivansteen/mc_rtc/build/tests/controllers/TestPostureController/mc_rtc-TestPostureController.conf" "400"
20: Test timeout computed to be: 1500
20: Running 1 test case...
20: [info] Loading default global configuration /usr/local/etc/mc_rtc.yaml
20: [info] Loading additional global configuration "/home/jarivansteen/.config/mc_rtc/mc_rtc.yaml"
20: [info] Loading additional global configuration /home/jarivansteen/mc_rtc/build/tests/controllers/TestPostureController/mc_rtc-TestPostureController.conf
20: [info] Enabled plugins: 
20: [info] GUI server enabled
20: [info] Will serve data on:
20: [info] - ipc:///tmp/mc_rtc_pub.ipc
20: [info] - tcp://*:4242
20: [info] Will handle requests on:
20: [info] - ipc:///tmp/mc_rtc_rep.ipc
20: [info] - tcp://*:4343
20: [info] Create controller TestPostureController
20: [warning] No calibration file /home/jarivansteen/catkin_data_ws/src/mc_rtc_data/jvrc_description/calib/jvrc1/calib_data.RightFootForceSensor found for force sensor RightFootForceSensor
20: [warning] No calibration file /home/jarivansteen/catkin_data_ws/src/mc_rtc_data/jvrc_description/calib/jvrc1/calib_data.LeftFootForceSensor found for force sensor LeftFootForceSensor
20: [warning] No calibration file /home/jarivansteen/catkin_data_ws/src/mc_rtc_data/jvrc_description/calib/jvrc1/calib_data.RightHandForceSensor found for force sensor RightHandForceSensor
20: [warning] No calibration file /home/jarivansteen/catkin_data_ws/src/mc_rtc_data/jvrc_description/calib/jvrc1/calib_data.LeftHandForceSensor found for force sensor LeftHandForceSensor
20: [warning] No calibration file /home/jarivansteen/catkin_data_ws/src/mc_rtc_data/jvrc_description/calib/jvrc1/calib_data.RightFootForceSensor found for force sensor RightFootForceSensor
20: [warning] No calibration file /home/jarivansteen/catkin_data_ws/src/mc_rtc_data/jvrc_description/calib/jvrc1/calib_data.LeftFootForceSensor found for force sensor LeftFootForceSensor
20: [warning] No calibration file /home/jarivansteen/catkin_data_ws/src/mc_rtc_data/jvrc_description/calib/jvrc1/calib_data.RightHandForceSensor found for force sensor RightHandForceSensor
20: [warning] No calibration file /home/jarivansteen/catkin_data_ws/src/mc_rtc_data/jvrc_description/calib/jvrc1/calib_data.LeftHandForceSensor found for force sensor LeftHandForceSensor
20: [info] MCController(base) ready
20: [info] Added task posture_jvrc1
20: [success] Created TestPostureController
20: [critical] [TestPostureController] The observer pipeline can no longer be configured by "EnabledObservers", "RunObservers" and "UpdateObservers".
20: Multiple pipelines are now supported, allowing for estimation of multiple robots and/or multiple observations of the same robot.
20: For details on upgrading, please refer to:
20: - The observer pipelines tutorial: https://jrl-umi3218.github.io/mc_rtc/tutorials/recipes/observers.html
20: - The JSON Schema documentation: https://jrl-umi3218.github.io/mc_rtc/json.html#Observers/ObserverPipelines
20/47 Test #20: TestPostureController ...............................***Exception: SegFault  0.35 sec
```

The two keys here are:

```
[info] Loading additional global configuration "/home/jarivansteen/.config/mc_rtc/mc_rtc.yaml"
```

so the controllers are loading local configuration (we should look at disabling this in the future)

and the actual failure:

```
20: [critical] [TestPostureController] The observer pipeline can no longer be configured by "EnabledObservers", "RunObservers" and "UpdateObservers".
20: Multiple pipelines are now supported, allowing for estimation of multiple robots and/or multiple observations of the same robot.
20: For details on upgrading, please refer to:
20: - The observer pipelines tutorial: https://jrl-umi3218.github.io/mc_rtc/tutorials/recipes/observers.html
20: - The JSON Schema documentation: https://jrl-umi3218.github.io/mc_rtc/json.html#Observers/ObserverPipelines
```

This indicates that you have some outdated pipeline configuration in your local configuration (`~/.config/mc_rtc/mc_rtc.yaml`). Once you remove those sections from your local configuration file the tests should run fine.

> I'm trying to install mc_rtc release 1.4.0 on Ubuntu 18.04.5 LTS

Strictly speaking, the script also update mc_rtc so you will be on 1.5.1 afterwards :-)

Feel free to close if this works out.

jari-van-steen, 2020-09-23:
Thanks @gergondet for the quick and useful response! I removed the local configuration file for the time being and tried building again, which did indeed resolve the issue!