jrl-umi3218/mc_rtc #77: Problems compiling an Observer

elfalko, 2020-09-30:
I'm currently struggling to compile an observer for an floating body.

<details>
<summary>source files</summary>
Folder structure
* mc_reemc/
|- * CMakeLists.txt
|- * src/
   |- * FloatingBaseObserver.h
   |- * FloatingBaseObserver.cpp
   |- ...
|- ...
```
</details>

arntanguy, 2020-10-01:
Thanks a lot for the very detailed issue.

In theory, your observer should have been automatically linked with `Eigen`, as the `add_observer` macro links the observer against `mc_rbdyn` which itself depends on `Eigen`. However this is indeed not the case. It seems that we might not be properly exporting the transitive dependency to the `Eigen` library, we'll look into it.

I have pushed a full sample repository with a working version of your example: https://github.com/arntanguy/mc_observer_example

A few notes:
```
cmake_minimum_required(VERSION 2.8.3)
```
This should be `3.1` minimum as `mc_rtc` uses modern target-based cmake.

```
add_observer(ReemcFloatingBaseObserver src/FloatingBaseObserver.cpp src/FloatingBaseObserver.h)
```
Since your observer inherits from `mc_observers::BodySensorObserver` and not from `mc_observers::Observer`, you need to link against the `BodySensorObserver`

```
target_link_libraries(ReemcFloatingBaseObserver PUBLIC mc_observers::BodySensorObserver)
```

I had however initially forgot to export those targets, this is now [fixed](https://github.com/jrl-umi3218/mc_rtc/commit/27c116d0a557406b42f29b70f2de0f0f010b5261) on mc_rtc master (sorry, you'll need to update).

For now, to link with `Eigen`, the following should work:

```
find_package(Eigen3 REQUIRED)

add_observer(ReemcFloatingBaseObserver src/FloatingBaseObserver.cpp src/FloatingBaseObserver.h)
target_link_libraries(ReemcFloatingBaseObserver PUBLIC mc_observers::BodySensorObserver)
if(TARGET Eigen3::Eigen)
  target_link_libraries(ReemcFloatingBaseObserver PUBLIC Eigen3::Eigen)
else()
  target_include_directories(ReemcFloatingBaseObserver SYSTEM "${EIGEN3_INCLUDE_DIR}")
endif()
target_include_directories(ReemcFloatingBaseObserver PUBLIC $<BUILD_INTERFACE:${CMAKE_CURRENT_SOURCE_DIR}> $<INSTALL_INTERFACE:include>)
```

gergondet, 2020-10-01:
> In theory, your observer should have been automatically linked with Eigen, as the add_observer macro links the observer against mc_rbdyn which itself depends on Eigen. However this is indeed not the [...]

We do it correctly (otherwise mc_rtc_ros would fail to build on Xenial)

[This is the issue](https://github.com/jrl-umi3218/mc_rtc/blob/master/src/mc_rtcMacros.cmake#L55), we should be linking with `mc_rtc::mc_rbdyn` and `mc_rtc::mc_rtc_gui` and in fact, we should be linking with `mc_rtc::mc_observers`. The non-namespaced targets do no exist when mc_rtc targets are imported, so even with the correct include path for Eigen, the build would have failed.

@elfalko you can copy/paste this macro in your project to get it to work:

```cmake
macro(add_observer observer_name observer_SRC observer_HDR)
  add_library(${observer_name} SHARED ${observer_SRC} ${observer_HDR})
  set_target_properties(${observer_name} PROPERTIES COMPILE_FLAGS "-DMC_OBSERVER_EXPORTS" PREFIX "")
  set_target_properties(${observer_name} PROPERTIES INSTALL_RPATH ${MC_OBSERVERS_INSTALL_PREFIX})
  target_link_libraries(${observer_name} PUBLIC mc_rtc::mc_observers)
  install(TARGETS ${observer_name} DESTINATION "${MC_OBSERVERS_INSTALL_PREFIX}")
endmacro()
```

elfalko, 2020-10-05:
Thanks for the responses! I tried the fixes on 
- a clean install with the package provided by @arntanguy (I also added the macro from @gergondet in a second run)
- my old system with my attempt at a package and your suggested fixes
- my updated old setup (also with fixes)

Unfortunately I keep getting the following error (of course with different package paths for the different setups):

``` bash
[...]
-- Found ROS libraries: /usr/lib/x86_64-linux-gnu/libboost_chrono.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_signals.so;/usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/liblog4cxx.so;/usr/lib/x86_64-linux-gnu/libboost_regex.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_atomic.so;/usr/lib/x86_64-linux-gnu/libpthread.so;/usr/lib/x86_64-linux-gnu/libconsole_bridge.so;/usr/lib/x86_64-linux-gnu/libboost_program_options.so;/usr/lib/x86_64-linux-gnu/liblz4.so;/opt/ros/kinetic/lib/libroscpp.so;/opt/ros/kinetic/lib/librosconsole.so;/opt/ros/kinetic/lib/librosconsole_log4cxx.so;/opt/ros/kinetic/lib/librosconsole_backend_interface.so;/opt/ros/kinetic/lib/libroscpp_serialization.so;/opt/ros/kinetic/lib/libxmlrpcpp.so;/opt/ros/kinetic/lib/librostime.so;/opt/ros/kinetic/lib/libcpp_common.so;/opt/ros/kinetic/lib/libtf2_ros.so;/opt/ros/kinetic/lib/libactionlib.so;/opt/ros/kinetic/lib/libmessage_filters.so;/opt/ros/kinetic/lib/libtf2.so;/opt/ros/kinetic/lib/librosbag.so;/opt/ros/kinetic/lib/librosbag_storage.so;/opt/ros/kinetic/lib/libroslz4.so;/opt/ros/kinetic/lib/libtopic_tools.so
-- Found ROS include directories: /opt/ros/kinetic/include
CMake Error at src/CMakeLists.txt:6 (target_include_directories):
  target_include_directories called with invalid arguments

  -- Configuring incomplete, errors occurred!
  See also "/home/f/devel3/catkin_ws/mc_observer_example/build/CMakeFiles/CMakeOutput.log".
  See also "/home/f/devel3/catkin_ws/mc_observer_example/build/CMakeFiles/CMakeError.log".
  make: *** No targets specified and no makefile found.  Stop.
```

arntanguy, 2020-10-06:
Hi,

Sorry, I had a small mistake for the non-target based Eigen inclusion (it should have been `  target_include_directories(ExampleFloatingBaseObserver SYSTEM INTERFACE "${EIGEN3_INCLUDE_DIR}")`). I just pushed a fix to the sample including Pierre's suggestion.
Note that the sample observer is not intended as a catkin package (it'd probably work with catkin too though).

gergondet, 2020-11-18:
Hi @elfalko 

The issue should be fixed in recent mc_rtc versions as the add_observer exported macros correctly links to the `mc_rtc::mc_observers` target. Can you confirm it works and (hopefully) close the issue?