jrl-umi3218/mc_rtc #89: build script fails

GAOXINCHI-20, 2020-11-17:
Hello,

The build_and_install.sh script fails with the following issue:
```
CMake Error at src/mc_robots/CMakeLists.txt:43 (generate_aliases):
  generate_aliases Macro invoked with incorrect arguments for macro named:
  generate_aliases

CMake Error at src/mc_robots/CMakeLists.txt:47 (generate_aliases):
  generate_aliases Macro invoked with incorrect arguments for macro named:
  generate_aliases
```
I am working on a fresh install of Ubuntu 16.04
Here is the log file:
[build_and_install_warnings-2020-11-17-15-50-31.log](https://github.com/jrl-umi3218/mc_rtc/files/5551778/build_and_install_warnings-2020-11-17-15-50-31.log)

Thanks in advance

gergondet, 2020-11-17:
Hi @GAOXINCHI-20 

From the log I can see:
```
-- Found mc_env_description: /usr/local
-- Found mc_int_obj_description: /usr/local
```

Whereas it should be (given your setup):
```
-- Found mc_env_description: /home/gaoxinchi/catkin_data_ws/src/mc_rtc_data/mc_env_description
```

Do you have files under `/usr/local/lib/cmake/mc_env_description`? If so please remove them and try to run the script again (same for `mc_int_obj_description` and `jvrc_description`)

I'm not sure why you would have those files though (especially from a fresh install...). Did you try a build without ROS support first? (although this should work as well)

GAOXINCHI-20, 2020-11-24:
Thanks for the answer, I successfully installed build_and_install.sh by reinstalling the UBUNTU system.

closed 2020-11-24 by GAOXINCHI-20