jrl-umi3218/mc_rtc #92: Computation time Spikes in realtime applications

Taherifar, 2020-11-18:
Hi,
I am writing to ask about the spikes (jitter) in the time consumption reflected in the perf object of the .bin file of the mc_rtc. I have implemented the mc_rtc and LIPM controller in 5ms sampling time in ROS. I have attached the screen shot of the perf function during the experiment with robot. It can be observed that there are spikes in which the computation time has increased to 15ms. These spikes might causes some difficulty in real-time data communication.
1- How can we prevent these spikes?
2- What have you done in your real-time applications, in the moments when the computation time is increased?

gergondet, 2020-11-19:
Hi @Taherifar 

The performances baseline seem pretty bad, for reference this is what I get on my computer which has a rather nice i7-7700. It's "spiky" when a state starts because this usually coincides with tasks/states allocations. We will be working on improving that in a future release as it can be problematic to run at high frequencies.

[image]

Few things to double-check:
- make sure `LogPolicy` is set to `threaded` in mc_rtc configuration, the default "non-threaded" policy can incur important spikes
- make sure all dependencies of lipm_walking controller and lipm_walking_controller are compiled with either `CMAKE_BUILD_TYPE=RelWithDebInfo` or `CMAKE_BUILD_TYPE=Release`
- make sure the CPU powersaving is disabled

> 2- What have you done in your real-time applications, in the moments when the computation time is increased?

Nothing special. The systems we worked with could handle iterations missing the deadline  if that happened rarely

Taherifar, 2020-11-20:
Thanks for your comments.
I applied all three points. It has great effect on reduction of spikes and also in the whole time consumption.

closed 2020-11-20 by Taherifar