jrl-umi3218/mc_rtc #97: [mc_log_ui] Robot menu and etc.

fkanehiro, 2020-12-02:
I'm running mc_rtc on ubuntu16.04.
After running my controller, I launched mc_log_ui.

1. Robot menu
Robot menu does not appear.
Is there any required condition to use Robot menu?

2. Common plots
When I choose "Common plots" -> "Commands" -> "OK", nothing happens.
Even if I choose "Select all" or "One plot per joint", nothing happens.
No message in the terminal where I launed mc_log_ui.
What is wrong?

gergondet, 2020-12-03:
Hi @fkanehiro 

1. The Robot menu only appears if you have (working) mc_rtc Python bindings, i.e. it only works if you can `import mc_rbdyn` with the python version used to run mc_log_ui (defaults to `python` i.e. `python2` in 16.04)

2. It might be the common plots only work properly if the `Robot` menu works, it should work regardless (just without nice names for the joints) so I will look into it. Normally when the window appear you should see a list of joint that you can check on/off (`Select all` simply checks everything on)

fkanehiro, 2020-12-03:
I thought Python binding would be enabled by WITH_PYTHON_SUPPORT, but it was not correct.
By adding PYTHON_FORCE_PYTHON2, both problems were solved.
Thank you for your comments and I think it is better to make "Common plots" work even if Python binding is disabled.

closed 2020-12-03 by fkanehiro