jrl-umi3218/mc_rtc #101: Issue with task configuration

L-P-P, 2020-12-07:
Hello,

I am having several issues configuring tasks for my application which I was hoping you might be able to resolve.
I am working on developing a controller for a bipedal robot using mc_rtc. This application involves the use of several tasks including the CoP, position, orientation and CoM tasks. However, the robot being employed has several redundant joints (causing each leg to have greater than 6 DOFs) and, for the purposes of my application, I would like to specify particular targets for these joints. Overall, I would like certain joints to follow a set of targets/trajectories very closely, while leaving the other joints free to accommodate the original tasks (the CoP, position and other tasks mentioned previously). I attempted to do this using the posture task where I specified targets only for the aforementioned redundant joints, and set the other joints to be inactive in regards to that task. I found in testing, however, that the joints specified in the posture task would not follow the given targets very closely, possibly being influenced by the other tasks. To resolve this, I set the weight and/or stiffness of this task to be higher but this caused the various other tasks to fail to meet their own targets acceptably. Setting the redundant joints to be inactive for the original tasks also did not resolve the issue. I also attempted to implement this with the orientation task, to control the orientation of the segments of the robot around the redundant joints and thus control the joints indirectly. I ran into issues with method as well since the orientation task constrains bodies with respect the world frame. Since the redundant joints are near the end-effector of each leg, and as a result of the kinematic structure between the relevant segment and the end-effector, constraining segments with respect to the world frame can prevent the CoP task from effectively rotating the end-effector to adapt to CoP deviations in particular directions, making it impractical for this application.

To summarize, I would like to control a robot through mc_rtc such that I can specify joint trajectories for a particular set of joints, while leaving the remaining joints to be used for a set of other tasks, but have encountered issues attempting to implement this with the mc_rtc posture or orientation tasks. Could you tell me if there are any tasks and/or constraints that might be implemented in mc_rtc that would allow this?

Thanks

gergondet, 2020-12-08:
Hi @L-P-P 

I think you have already tried some of the possible solutions but I think the following steps could work, let's call the joints you want to control via the posture task the "extra joints" `["J0", ..., "JN"]`:

1. [Apply joint selection](https://jrl-umi3218.github.io/mc_rtc/tutorials/recipes/joint-select.html) for the tasks that shouldn't use your extra joints, e.g. `comTask->selectUnactiveJoints(solver(), {"J0", ..., "JN"})`
2. [Apply high joint stiffness](https://jrl-umi3218.github.io/mc_rtc/tutorials/recipes/joint-stiffness.html) to the extra joints, e.g. `postureTask->jointStiffness(solver(), {{"J0", 1e3}, ..., {"JN", 1e3}})` while keeping a low weight/stiffness on the posture task (e.g. 1/1, in particular do not set the stiffness to 0)

L-P-P, 2020-12-09:
Thank you for your recommendations. Steps 1 and 2 together resolved the issue sufficiently.

closed 2020-12-09 by L-P-P