jrl-umi3218/mc_rtc #102: Issue with mc_rtc_new_controller: git-python on Ubuntu 20.04

alecksbigbag, 2020-12-07:
Following the instructions 

   https://jrl-umi3218.github.io/mc_rtc/tutorials/introduction/first-controller.html

I found the comment  

`Note: this tool requires Git for Python which is available as python-git in Debian-like systems and GitPython in pip`

However, as I try to install Git for Python on Ubuntu 20.04 using `sudo apt install python-git` I run into the following issue

```bash
$ sudo apt install python-git
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package python-git is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python-git-doc

E: Package 'python-git' has no installation candidate
```

So my question is, can I actually run mc_rtc_new_controller on Ubuntu 20.04?
I am not an expert, but it seems such a package is not available looking here
 
     https://packages.ubuntu.com/search?suite=focal&keywords=python-git

Thanks.

gergondet, 2020-12-08:
Hi Alessandro,

Indeed this suffers the same issue as `mc_log_ui` in Ubuntu 20.04 (which has been fixed in #94). I'll get around to fix this.

In the meantime you can either:
1. `sudo apt install python-is-python3 python3-git`
2. Use the new [template project](https://github.com/mc-rtc/new-controller)/[FSM template project](https://github.com/mc-rtc/new-fsm-controller) (this is isn't listed on the tutorials yet but will be once #88 gets merged)

alecksbigbag, 2020-12-08:
> Reply to: jrl-umi3218/mc_rtc <reply@reply.github.com>

Hi Pierre,
thank you. Clear.

Alessandro

closed 2020-12-09 by gergondet