jrl-umi3218/mc_rtc #111: Issue with JVRC1 robot model

Saeed-Mansouri, 2020-12-31:
I have run the LIPM controller with the JVRC1 robot model. As you can see in the below image, the robot leaned to the left during the forward walking motion. I have run this package with my own robot model, and I haven't seen this issue. Have you seen such an issue with the JVRC1 robot model?

[image]

gergondet, 2021-01-04:
Hi @Saeed-Mansouri 

Thanks for reporting this issue. We have never tuned the stabilizer settings for JVRC1 nor really looked closely at the lipm_walking_controller performance, we simply got it to work well enough to run in integration tests.

In what environment did you run these simulations?

Saeed-Mansouri, 2021-01-04:
I ran these simulations in both mc_rtc_ticker and Choreonoid environments, and the results were the same.

closed 2023-09-04 by gergondet