jrl-umi3218/mc_rtc #118: [gui] Change polygon line width

mmurooka, 2021-01-18:
This is a light request regarding GUI.

LineConfig (including not only color but also width) can be passed to Trajectory constructor: https://github.com/jrl-umi3218/mc_rtc/blob/e383426418669292fcb9eea58b26de3e3a22c0fa/include/mc_rtc/gui/Trajectory.h#L67
But only color can be passed to Polygon constructor.
https://github.com/jrl-umi3218/mc_rtc/blob/e383426418669292fcb9eea58b26de3e3a22c0fa/include/mc_rtc/gui/Polygon.h#L66

Is it possible to make the Polygon constructor also receive LineConfig?

I'd like to make the polygon line of the footstep thicker in the videos in https://gite.lirmm.fr/jrp-michelin/mc_hwm_planner/-/merge_requests/1 (only for internal).

gergondet, 2021-01-29:
Hi @mmurooka 

Thanks for the suggestion. It makes sense indeed. Note that this requires https://github.com/jrl-umi3218/mc_rtc_ros/commit/e09f932c3b3bfc5cbf2d3ddaf20abc314c44a88d to be handled on the client side

Note: the RViZ client does not handle dotted lines for now

closed 2021-01-29 by gergondet

mmurooka, 2021-01-29:
Thank you @gergondet !

commit 4aff8912 (2021-01-29): [GUI] Allow to pass a LineConfig for Polygon elements

- Update PROTOCOL_VERSION since old client cannot handle the
configuration correctly
- *but* make the change backward compatible for clients implementations

Close #118