jrl-umi3218/mc_rtc #120: Rotation transformation seems incorrect in stabilizer

mmurooka, 2021-01-30:
It seems that the coordinate transformation associated with the Yaw rotation of the robot is not handled correctly in (maybe) StabilizerTask.

Watching the video below is probably the fastest way to share the issue: https://www.dropbox.com/s/duo4vy51wvrsqlt/mc_rtc-stabilizer-transformation-bug-20210130.mp4?dl=0

In the video, HRP2 is doing the 1m forwards walking with lipm_walking_controller twice. In the first time, the robot is placed in the default attitude, and in the second time, the configure file (yaml file) is edited and the robot is placed with rotated 90 degrees in yaw. Only during the second walk, there is a large error between the control robot and the real robot.
(A slight change has been made to lipm_walking_controller to reflect the initial robot position in the configure file: https://github.com/mmurooka/lipm_walking_controller/commit/3d882da2eee25dfddc2104894b78755e88043d08.)

The following is graphs of the first and second walks. The second time the robot is almost falling.
[image]
[image]

Some notes:
- This problem occurs not only with lipm_walking_controller but also with [my controller](https://gite.lirmm.fr/jrp-michelin/mc_hwm_planner). Different methods are used to generate the CoM and swing trajectory, and the main common point is that the StabilizerTask is used. (So, I created an Issue in this repository.)
- Instead of rotating the initial placement 90 degrees, if the robot makes a 90 degree turn after placing it in the default attitude, the same problem occurs when walking forward.
- In the video, I tried it with mc_rtc_ticker, but when I tried it with choreonoid, the robot falls down during the second walk.

mmurooka, 2021-01-30:
One more important note:
- This problem was confirmed on HRP2, but was not reproduced on JVRC1 and HRP5P. In JVRC1 and HRP5P, the graphs were completely the same in two forward walks (with the default initial attitude and with the initial attitude rotated 90 degrees).

arntanguy, 2021-02-01:
Thanks for pointing this out!
I'll look into it.

arntanguy, 2021-02-01:
It seems there was a bug in the kinematic inertial observer. #121 should fix the issue.

commit 1ce4c07b (2020-05-06): Merge branch 'topic/MiscFixes' into 'master'

Misc fixes

Closes #120, #148, #153, #171, #205, and #106

See merge request multi-contact/mc_rtc!213
commit 8f35c5ef (2020-05-06): [mc_log_ui] Change font type so that the generated figures pass RAS PDF check

Closes #120 (GitLab)
closed 2021-07-13 by gergondet