jrl-umi3218/mc_rtc #133: [plugin] ROS configuration is not applied

mmurooka, 2021-03-11:
I put the ROS plugin configuration according to [tutorial](https://jrl-umi3218.github.io/mc_rtc/tutorials/introduction/configuration.html#possible-locations) as follows, but it is not applied correctly:
```bash
$ ~/.config/mc_rtc/plugins/ROS.yaml
# Control which data we publish and publishing rate
publish:
  # Timestep of publication (ROS)
    timestep: 0.03
```

When running the sample CoM controller with [the branch](https://github.com/jrl-umi3218/mc_rtc/compare/master...mmurooka:debug-publisher-rate), the output becomes as follows:

```bash
$ choreonoid sim_mc.cnoid --start-simulaiton

...

[info] MCControl::onInitialize() starting
[info] MCControl::onInitialize() finished
KalmanFilter::onActivated(1000)
KalmanFilter: Q = 0.01, Q_bw = 0.001, R = 0.1, dt = 0.005, filter order = 0, Tgsens = 0.05
[info] onActivated
[success] publisher: control, impl.publish_rate: 100.0
[success] publisher: control/env_1, impl.publish_rate: 100.0
[success] publisher: real/env_1, impl.publish_rate: 100.0
[success] publisher: real, impl.publish_rate: 100.0
[info] Init controller
[info] Initializing attitude from robot module: q=[1.000, 0.000, 0.000, 0.000, 0.000, 0.000, 0.828]
[info] Will log controller outputs to /tmp/mc-control-CoM-2021-03-11-16-17-35.bin
[info] Updated latest log symlink: /tmp/mc-control-CoM-latest.bin
[info] Added task com_jvrc1
[success] publisher: PublisherMadeInReset, impl.publish_rate: 100.0
[success] [mc_rtc::ROS] Starting ROS services
[success] publisher: PublisherMadeInRun, impl.publish_rate: 33.333333333333336
```

Please see the line start with `[success] publisher:`.
Only the publisher `PublisherMadeInReset` made in `Controller::run` (i.e., `mc_rtc::ROSBridge::update_robot_publisher` is called for the first time in `Controller::run`) reflects the specified timestep (0.03 sec = 33.333 Hz). For the default publishers (e.g., `control`) and the publisher made in `Controller::reset`, the publish rate is not changed from the default (100.0 Hz).

gergondet, 2021-03-11:
Hi @mmurooka 

This is a bit of a chicken and egg problem. The plugins are initialized (and thus configured) or reset after the controller. That's why the publisher that is created in the reset function has the default publishing rate but the one created in the run loop has the user-provided one.

However, this doesn't affect the publishers created by the plugin which have the correct timestep. (I've checked now, if I add a configuration file like yours then `rostopic hz /control/imu` gives me 33.33Hz)

mmurooka, 2021-03-15:
@gergondet, thank you for investigation and reply.

The configuration of ROS publishing rate was reflected correctly when launching by `mc_rtc_ticker` as you pointed out. But it was not reflected when launching by `choreonoid` + `mc_openrtm`.

Followings are the result with the ROS plugin configuration in https://github.com/jrl-umi3218/mc_rtc/issues/133#issue-828863974 (i.e., `timestep: 0.03`).

### case 1 (mc_rtc_ticker)
```
$ rosrun mc_rtc_ticker mc_rtc_ticker
$ rostopic hz /control/imu -w 100
subscribed to [/control/imu]
average rate: 33.336
	min: 0.030s max: 0.030s std dev: 0.00012s window: 33
average rate: 33.334
	min: 0.030s max: 0.030s std dev: 0.00011s window: 66
average rate: 33.333
	min: 0.030s max: 0.030s std dev: 0.00011s window: 100
```

### case2 (mc_openrtm)
```
$ choreonoid sim_mc.cnoid --start-simulation
$ rostopic hz /control/imu -w 100
subscribed to [/control/imu]
average rate: 99.995
	min: 0.010s max: 0.010s std dev: 0.00013s window: 95
average rate: 99.988
	min: 0.000s max: 0.020s std dev: 0.00201s window: 100
average rate: 100.001
	min: 0.000s max: 0.020s std dev: 0.00245s window: 100
```

gergondet, 2021-03-15:
Hi @mmurooka 

Thanks for reporting. The issue was that in mc_openrtm case, the plugin can run before the controller is reset as it runs an "empty" loop before the controller gets started.

The above commit allows to reset the publishing rate of a robot publisher and thus the publishers gets the correct publishing rate regardless of when they get initialized (or if the publishing rate gets modified at runtime through another way)

closed 2021-03-15 by gergondet

mmurooka, 2021-03-24:
@gergondet, I confirmed that it works correctly, thank you!

commit 0d5861f5 (2021-03-15): [ROS] Allow to reset RobotPublisher publication rate

This should fix #133