jrl-umi3218/mc_rtc #134: [installation] pip is called before installed

mmurooka, 2021-03-12:
Sorry for missing a log file, but when I was running the mc_rtc installation script on clean installed Ubuntu 18.04, I saw a error of `pip command not found`.

This is because the line
https://github.com/jrl-umi3218/mc_rtc/blob/96eed4d78aa5bcfb0d0eadab30bb5c7735d6a307/utils/build_and_install.sh#L532
is called before installing depent commands line
https://github.com/jrl-umi3218/mc_rtc/blob/96eed4d78aa5bcfb0d0eadab30bb5c7735d6a307/utils/build_and_install.sh#L605

N1ckfm, 2021-03-17:
Hi, did you try to install pip?
`sudo apt install python3-pip`

mmurooka, 2021-03-17:
@N1ckfm, thanks for the comment. Yes, I solved this issue by manually installing pip. (However, I don't think it is ideal as an installation script to handle errors manually, so this issue is for future improvement.)

gergondet, 2021-03-18:
Hi @mmurooka 

Thanks for reporting the issue. The python/pip coherency is now checked after system dependencies are installed.

commit ccb1b856 (2021-03-18): [utils] Check python/pip coherency after installing system dependencies

Closes #134
closed 2021-03-18 by gergondet