jrl-umi3218/mc_rtc #136: Could not find a configuration file for package "mc_env_description" that is compatible with requested version

N1ckfm, 2021-03-14:
Hi,  I will appreciate your help
when I follow the tutorial in the step 6: 
[https://jrl-umi3218.github.io/mc_rtc/tutorials/introduction/installation-guide.html](https://jrl-umi3218.github.io/mc_rtc/tutorials/introduction/installation-guide.html) I got a CMake error:
_Could not find a configuration file for package "mc_env_description" that
  is compatible with requested version ""._

OS: Ubuntu 20.04.2 LTS
ROS version: Noetic

```
Run ./build_and_install.sh
```
My result when building mc_rtc:

```
=====================
== Building mc_rtc ==
=====================

Extrayendo origin
fatal: no es posible acceder 'https://github.com/jrl-umi3218/mc_rtc/':Could not resolve host: github.com
error: No se pudo extraer origin
-- [WARNING] Would compile mc_rtc from commit 2bc6f2dc3bfb9030ae09da1b6bfa942d6abb7a90 (currently on branch master) but the remote master branch is at fa95f17b74c406ad0428e1901fcce85d06295d6c
Are you sure? [y/N] 
```
if I choice y, I got this:
```
-- Building mc_rtc from commit 2bc6f2dc3bfb9030ae09da1b6bfa942d6abb7a90
Sincronizando url del submódulo para '.jrl-ci'
Sincronizando url del submódulo para 'cmake'
+ cmake /home/nicolasfm/mc_rtc -DCMAKE_BUILD_TYPE:STRING=RelWithDebInfo -DCMAKE_INSTALL_PREFIX:STRING=/usr/local -DBUILD_TESTING:BOOL=ON -DBUILD_BENCHMARKS:BOOL=OFF -DPYTHON_BINDING:BOOL=ON -DPYTHON_BINDING_USER_INSTALL:BOOL=OFF -DPYTHON_BINDING_FORCE_PYTHON2:BOOL=OFF -DPYTHON_BINDING_FORCE_PYTHON3:BOOL=ON -DPYTHON_BINDING_BUILD_PYTHON2_AND_PYTHON3:BOOL=OFF -DBUILD_PYTHON_INTERFACE:BOOL=OFF -DSPDLOG_BUILD_EXAMPLE:BOOL=OFF -DSPDLOG_BUILD_SHARED:BOOL=ON -DDISABLE_ROS=OFF (pwd: /home/nicolasfm/mc_rtc/build)
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: filesystem timer 
-- Tasks FOUND. Tasks at /usr/local/lib/libTasks.so
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: system 
-- sch-core searched on  FOUND. sch-core at /usr/local/lib/libsch-core.so
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: serialization 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: timer 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: system 
-- state-observation FOUND. state-observation at /usr/local/lib/libstate-observation.so
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: timer 
-- Found hpp-spline include directories: /usr/local/include
-- Found GEOS by geos-config: /usr/lib/x86_64-linux-gnu/libgeos.so
-- Found nanomsg libraries: /usr/lib/x86_64-linux-gnu/libnanomsg.so
-- 
-- ========== qhull Build Information ==========
-- Build Version:                             8.0.2
-- Install Prefix (CMAKE_INSTALL_PREFIX):     /usr/local
-- Binary Directory (BIN_INSTALL_DIR):        bin
-- Library Directory (LIB_INSTALL_DIR):       lib
-- Include Directory (INCLUDE_INSTALL_DIR):   include
-- Documentation Directory (DOC_INSTALL_DIR): share/doc/qhull
-- Man Pages Directory (MAN_INSTALL_DIR):     share/man/man1
-- Build Type (CMAKE_BUILD_TYPE):             RelWithDebInfo
-- Build static libraries:                    ON
-- Build shared library:                      ON
-- Use shared library for linking apps:       OFF
-- To override these options, add -D{OPTION_NAME}=... to the cmake command
--   Build the debug targets                  -DCMAKE_BUILD_TYPE=Debug
-- 
-- To build and install qhull, enter "make" and "make install"
-- To smoketest qhull, enter "ctest"
-- 
CMake Error at CMakeLists.txt:111 (find_package):
  Could not find a configuration file for package "mc_env_description" that
  is compatible with requested version "".

  The following configuration files were considered but not accepted:

    /usr/lib/cmake/mc_env_description/mc_env_descriptionConfig.cmake, version: 1.0.4 (32bit)

Call Stack (most recent call first):
  CMakeLists.txt:127 (find_description_package)

-- Configuring incomplete, errors occurred!
See also "/home/nicolasfm/mc_rtc/build/CMakeFiles/CMakeOutput.log".
See also "/home/nicolasfm/mc_rtc/build/CMakeFiles/CMakeError.log".
CMake configuration failed for mc_rtc
Installation failed.
Installation log has been written to /tmp/build_and_install_warnings-2021-03-13-20-30-33.log
Please fix the issues and re-run the script.
```

N1ckfm, 2021-03-14:
[CMakeError.log](https://github.com/jrl-umi3218/mc_rtc/files/6135941/CMakeError.log)
[CMakeOutput.log](https://github.com/jrl-umi3218/mc_rtc/files/6135942/CMakeOutput.log)

gergondet, 2021-03-15:
Hi @N1ckfm 

The issue is from:

```
  The following configuration files were considered but not accepted:

    /usr/lib/cmake/mc_env_description/mc_env_descriptionConfig.cmake, version: 1.0.4 (32bit)
```

I'm not sure how you ended up with a 32 bits installation of this package (especially on Ubuntu 20.04). The installation script does not install anything in `/usr` and the Ubuntu 20.04 packages should only exist in 64 bits version.

Did you try to install binary packages before going with the "Building from source" road on the installation guide? If so, I'd advise you remove the packages that were installed during that step and only run the build_and_install script.

To confirm this hypothesis, can you check the output of `dpkg -l|grep mc-rtc-data`, it should look like below, but I'm assuming you have a different version somehow.
```bash
$ dpkg -l|grep mc-rtc-data
ii  mc-rtc-data                            1.0.4-1ubuntu1-20200923141856-d5868f51     all          Environments/Robots description for mc_rtc
ii  ros-noetic-mc-rtc-data                 1.0.4-1ubuntu1-20200923141856-d5868f51     all          Environments/Robots description for mc_rtc (ROS)
```

N1ckfm, 2021-03-16:
hi @gergondet I removed the 32 bit architecture, but i was still having problems.
I do not know if it has to do but I had installed ubuntu 16.04 and upgraded to ubuntu 20.04. 
So this time I removed ubuntu and did a fresh install of ubuntu 20.04 LTS, and was able to install mc_rtc with no problem.

Thanks for your support
Nícolas

closed 2021-03-16 by N1ckfm