jrl-umi3218/mc_rtc #137: Running the controller

N1ckfm, 2021-03-17:
hi, I'm having trouble loading "MyFirstController" I would like your support:
This is the tutorial:
https://jrl-umi3218.github.io/mc_rtc/tutorials/introduction/first-controller.html

```
mc_rtc_new_controller my_first_controller MyFirstController
mkdir -p build
cd build
cmake ../ -DCMAKE_BUILD_TYPE=RelWithDebInfo
make
sudo make install
```

When mentioned: _"Modify your mc_rtc configuration file"_
```
{
  "MainRobot": "JVRC1",
  "Enabled": ["MyFirstController"]
}
```
Refers to the following file?
`/my_first_controller/etc/MyFirstController.in.yaml`

When mentioned: _"Then run your controller as explained in the previous section. "_
Refer to this link?
https://jrl-umi3218.github.io/mc_rtc/tutorials/introduction/running-a-controller.html

It means that I have to execute the same code: 
```
roslaunch mc_rtc_ticker display.launch
rosrun mc_rtc_ticker mc_rtc_ticker
```
And should I see in the simulation the option to choose myfirstcontroller? (because i cant see it)
Or I should the change some part _rosrun myfirstcontroller.._  
I will appreciate your help
Thanks!
Nícolas

gergondet, 2021-03-17:
Hi @N1ckfm 

This is referring to mc_rtc configuration file as described [here](https://jrl-umi3218.github.io/mc_rtc/tutorials/introduction/configuration.html)

On Linux, create a local mc_rtc configuration folder:
```bash
mkdir -p ~/.config/mc_rtc/
```

Then add the content into `~/.config/mc_rtc/mc_rtc.yaml`:
```yaml
MainRobot: JVRC1
Enabled: MyFirstController
```

Finally, run it:
```bash
roslaunch mc_rtc_ticker display.launch
rosrun mc_rtc_ticker mc_rtc_ticker
```

closed 2021-03-17 by N1ckfm

Aimiliospet, 2022-05-13:
hello, sorry I'm also having trouble with running **"MyFirstController"**. Im getting the error "no controller enabled" when executing "**roslaunch mc_rtc_ticker display.launch"** and then **"rosrun mc_rtc_ticker mc_rtc_ticker",** although i have implemented all the steps from the tutorial as they were stated. I created the **my_first_controller** package with the "mc_rtc_new_controller" script inside the **mc_rtc/src/mc_control** folder. Should it be located elsewhere or isn't this the problem ? Am I missing smth else? 
Thank you in advance!
Aimilios