jrl-umi3218/mc_rtc #142: Issue Moving the CoM up and down

N1ckfm, 2021-04-01:
I am following the tutorial: [https://jrl-umi3218.github.io/mc_rtc/tutorials/introduction/com-controller.html](https://jrl-umi3218.github.io/mc_rtc/tutorials/introduction/com-controller.html)
When I use the following code:
[https://jrl-umi3218.github.io/mc_rtc/tutorials/introduction/sources/com-controller.html](https://jrl-umi3218.github.io/mc_rtc/tutorials/introduction/sources/com-controller.html)

and I run the following:
```
roslaunch mc_rtc_ticker display.launch
rosrun mc_rtc_ticker mc_rtc_ticker
```
the robot only moves its head but not moving up and down.
I will appreciate your support
Thank!

Nícolas

gergondet, 2021-04-01:
Hi @N1ckfm 

Did you compile and installed the modified code? The console of `rosrun mc_rtc_ticker mc_rtc_ticker` should say something about a `CoMTask` being added to the controller

N1ckfm, 2021-04-01:
Hi @gergondet  yes, I needed to compiled and installed:
```
$ cd build
# This build type provides good performance with debuggable code
$ cmake ../ -DCMAKE_BUILD_TYPE=RelWithDebInfo
$ make
$ sudo make install
```
I got the robot to move up and down but I get this error:
`[error] PostureTask :: target dof missmatch for NECK_Y`
I will appreciate your comments

N1ckfm, 2021-04-02:
I found the solution!
in this link `https://jrl-umi3218.github.io/mc_rtc/tutorials/introduction/sources/com-controller.html`
the MyFirstController.cpp is missing a code line before of `config_.load(config);`
`jointIndex = robot().jointIndexByName("NECK_Y");`

closed 2021-04-02 by N1ckfm