jrl-umi3218/mc_rtc #145: mc_rbdyn cannot link to sch::S_Cylinder

TsuruMasato, 2021-04-12:
Hello,

I met a problem in linking mc_rbdyn and sch-core.

When I checked the linking relationships by using the below command :

`$ ldd -r /usr/local/lib/libmc_rbdyn.so.1.7.0`

I saw a linker error as :

`undefined symbol: sch::S_Cylinder::S_Cylinder(CD_Matrix::Vector3T<double, false>, CD_Matrix::Vector3T<double, false>, double)  (/usr/local/lib/libmc_rbdyn.so.1.7.0)`

I am using the nice installation script to establish my mc_rc environment, and the script file finished correctly.
so sch-core has been compiled successfully.
Do you have any idea?

Thank you.

gergondet, 2021-04-12:
Hi @TsuruMasato 

Thanks for reporting the issue.

> I am using the nice installation script to establish my mc_rc environment, and the script file finished correctly.

Since the script is completing successfully could you explain where you you encounter a linking error?

TsuruMasato, 2021-04-12:
Thank you for replying, Dr.Gergondet.
I found this issue when I am compiling mc_udp.

My terminal complained about the definition of a constructor _sch::S_Cylinder::S_Cylinder_ , so that I checked libmc_rbdyn.so.1.7.0.

[image]

TsuruMasato, 2021-04-12:
My compiler is now using sch in openrtp/lib/libsch-core.so.
I guess mc_rbdyn should be connected with /usr/lib/libsch-core.so instead, shouldn't it ?

If so, I will set the configuration to declare the sch's place.

[image]

mmurooka, 2021-04-12:
@TsuruMasato,

In my local environment, `libmc_rbdyn.so` is linking to `libsch-core.so` under the `openrtp` directory like your case. But there is no warning about undefined symbol `S_Cylinder`. 
```
$ ldd -r libmc_rbdyn.so | grep sch
	libsch-core.so.1 => /home/mmurooka/openrtp/lib/libsch-core.so.1 (0x00007f433b80b000)
```

Isn't the source revision of `sch-core` (the one set up by the drcutil script) installed to `openrtp` out of date? If its remote is `mehdi-benallegue/sch-core`, you need to switch to `jrl-umi3218/sch-core` and `git pull`.
ref: https://github.com/jenkinshrg/drcutil/pull/50

gergondet, 2021-04-13:
Hi @TsuruMasato 

I think you have two installs of sch-core (one in `/usr/local` and one in `$HOME/openrtp`)

During the script, the prefix is favored for searching libraries so mc_rtc will pick up on the newly installed libraries in `/usr/local` but I'm guessing in your environement `$HOME/openrtp/lib` comes first and thus the one in `$HOME/openrtp` gets picked up first.

As @mmurooka pointed out, changing the remote on the sch-core repository used by the drcutil script will fix the discrepancy between the two versions (`sch::S_Cylinder` is relatively recent and probably not present in 1.0.3)

TsuruMasato, 2021-04-13:
OK, thank you so much...!!
I will try it.

TsuruMasato, 2021-04-14:
Thank you Pierre-san and Murooka-san!!
I fixed it!!

1. I confirmed the sch-core's repository and branch as jrl-umi3218/master first, then manually re-built (cmake&make&install) ech-core and mc_rtc cleanly, even though it didn't connect...
So actually the source code was already the latest version.

2. So next, I updated the whole openrtp by using its script file first, then I used mc_rtc's script file and it connected mc_rbdyn and ecs-core correctly.
Now mc_rbdyn is linking with /usr/local/lib/libsch-core.so, it has automatically changed... : )

In conclusion,

As your point out, I had two sch-core in my computer and mc_rtc had linked with sch-core.so at /openrtp/lib before.

So the fastest solution must be, to change CMAKE_INSTALL_PREFIX as /$home/openrtp/lib , at the step 1.

(After fixing everything, I noticed that  /openrtp/lib/sch-core.so had not been updated for a long time.)

Anyway, thanks to your advice, I fixed the problem!
Thank you so much for your quick responses!!

mmurooka, 2021-04-14:
The following is just a notice, so no response is needed.

> On the other hand, my re-build process updated only /usr/local/libsch-core.so.

By executing [drcutil script](https://github.com/jenkinshrg/drcutil), the `libsch-core.so` in `$HOME/openrtp` directory should be updated, and by executing [mc_rtc script](https://github.com/jrl-umi3218/mc_rtc/tree/master/utils), the `libsch-core.so` in `/usr/local` is updated.

In my guess, you only checked and updated the source code in the workspace setup by mc_rtc script in the following first process (Here, the version of the sch-core source cloned by the drcutil script was important)

> First, I confirmed the sch-core's repository and branch as jrl-umi3218/master, then manually re-built (cmake&make&install) ech-core and mc_rtc cleanly, but mc_rbdyn didn't work correctly...
I mean, the source code was actually already the right and latest version.

But anyway, you also have updated `openrtp` workspace in the second process, so it is up to date

> So next, I updated the whole openrtp by using its script file.
Then I used mc_rtc's script file and it connected mc_rbdyn and ech-core correctly.
Now my mc_rbdyn.so is linking with /usr/local/lib/libsch-core.so. it has automatically changed... : )

TsuruMasato, 2021-07-30:
Sorry again.

I am facing the same issue now.
But this time, I cannot fix it.

[image]

When I run mc_ana_avatar_contorller, it failed to load MCControl.so. (Left terminal in the image)
The reason is a linker error between mc_rbdyn and sch-core, again. (Right terimnal in the image)

I have tried to fix this linker error a lot, but I cannot.
Does anyone know how to fix it?

(p.s. sch-core is jrl-umi3218/master, and mc_rtc is the latest version...)

gergondet, 2021-08-02:
Hi @TsuruMasato 

I'm guessing you have one version of sch-core in `$HOME/openrtp/lib` and a different version in `/usr/local/lib` which could lead to ssuch errors.

TsuruMasato, 2021-08-02:
Finally I could find the place of definition of sch-core place and  fixed it.

closed 2021-08-02 by TsuruMasato