jrl-umi3218/mc_rtc #146: [Plugin] before() and after() functions runs once before finishing init()

TsuruMasato, 2021-04-19:
Dear Dr.Gergondet @gergondet ,
(cc: @chappk )

This is Tsuru, JRL.
Now I am creating a new mc_plugin, and I found an issue in Global Plugin.

There are 4 functions, init(), reset(), before(), and after() in Global Plugin and its inheritance classes.

As far as I checked, before() and after() functions start their process without waiting init() function.

On the attached screen shot, there is a debug message 
`[info] [FootstepPlugin] before` 
first.

After that, second debug message 
`[info] [FootstepPlugin] after` 
came soon.

And finally, there is a sequence of debug messages written in my init() function, as 
`[success] [FootstepPlugin] start init`, 
`[success] [FootstepPlugin] success in getting ROS nh`, and  
`[success] [FootstepPlugin] success init()`.

[image]

Originally, Kevin and I were understanding that 
init() runs first, then before() and after() start with the mc_rtc's main loop.

Is it an expected process for you?

Thank you.

P.S. This plugin is still a draft, and there are only a few, simple datastore reading/writing functions in init(), before(), and after().

gergondet, 2021-04-26:
Hi @TsuruMasato 

This is a normal behavior at the moment. It is possible to call `GlobalController::run()` before calling `GlobalController::init()`, this is specifically intended for plugins to run before the controller can start. (I think this is only used in the OpenRTM interface though)

I have been thinking about improving the global plugin system so that we have a finer control over this behavior, in particular:
- allow a plugin to specify whether it implements only one of `run_before`/`run_after` or even neither (the later can be useful to have a plugin that only imports types into the task or constraint loader)
- allow a plugin to specify that it cannot/should not run before the `init` call
- add a `reset` function to the plugin so that it can adapt (if needed) when the controller changes (or right after the `init` call)

Does that sound good to you? Would you need something else in the plugin system?

TsuruMasato, 2021-04-26:
Thank you for your reply, Dr.Gergondet.

OK, if it is a normal behavior, not so big problem. Just I was an amateur about mc_rtc.
Exactly, I was running my plugin with Choreonoid simulation.

About your 3 ideas, I think they make plugin system more natural? and easier for users to understand.

Additionally, I wonder there is a sample plugin code and guidance on the official website.
I have written several mc_plugins in my last paper, so I might contribute to it.

gergondet, 2021-07-29:
Hi @TsuruMasato 

This has been implemented in #168 and now plugins can configure themselves how/when they should run

closed 2021-07-29 by gergondet