jrl-umi3218/mc_rtc #147: Replace mc_rtc_rviz_panel commands with another client to run Choreonoid

Ahmadreza-Shahrokhshahi, 2021-04-24:
Hi,

Instead of using _mc_rtc_rviz_panel_ to send some requests through GUI, I am trying to use a python script to send the commands to the mc_rtc server and eventually to the controller. I used both telnet and my python script to connect to the server on the localhost at ports 4242 and 4343. The connection was established successfully. Then I sent this string as the start command to the server "{"category":["Walking","Main"],"name":"Start standing","data":{}}" which I know is being sent by the _mc_rtc_rviz_panel_.

What I received as a response on the client-side is " SP " with some other characters which are not shown here. And the connection was forced to the closed state as soon as I sent the data.

This is what I receive when I try to connect to port 4242:
[image]

and this is what I receive when I try to connect to port 4343:
[image]

Now the question is am I missing something here? AFAIK, port 4343 is for sending requests. But when I send the requests nothing is being received on the server-side.

What is the best way to send controller commands and requests to the mc_rtc server without using GUI in Python?
Thanks

gergondet, 2021-04-26:
Hi @Ahmadreza-Shahrokhshahi 

mc_rtc does not use raw socket for transporting messages between the GUI server and the client. It uses nanomsg to do so.

More precisely:
- the server sends the GUI status via a [pub/sub socket](https://nanomsg.org/gettingstarted/pubsub.html)
- the client sends commands to the server via [pipeline socket](https://nanomsg.org/gettingstarted/pipeline.html)

There is various [nanomsg bindings for Python](https://nanomsg.org/documentation.html) if you want to give it a try to send requests yourself. The more future proof approach would be to implement a Python binding for the `mc_control::ControllerClient` class.

It is more difficult for correctly receiving the message from the GUI because the message is encoded via [mpack](https://github.com/ludocode/mpack) and the `mc_control::ControllerClient` class takes care of decoding this message correctly.

gergondet, 2021-08-17:
Hi @Ahmadreza-Shahrokhshahi 

I'm going to close this issue. Feel free to re-open if you have more questions on this topic.

closed 2021-08-17 by gergondet