jrl-umi3218/mc_rtc #149: Tutorial FSM controller in practice. No controller found

luigipenco, 2021-04-25:
Following the tutorial "FSM controller in practice", after creating the controller with the command
mc_rtc_new_fsm_controller my_first_fsm_controller MyFirstFSMController  , 
building and running it as specified,
the controller is not found. The confi file ~/.config/mc_rtc/mc_rtc.yaml has the right controller name, so I think the library is not found or properly linked.

[warning] Controller MyFirstFSMController enabled in configuration but not available
[error] No controller selected or selected controller is not enabled, please check your configuration file
[info] Note common reasons for this error include:
	- The controller name does not match the name exported by CONTROLLER_CONSTRUCTOR
	- The controller library is not in a path read by mc_rtc
	- The controller constuctor segfaults
	- The controller library hasn't been properly linked

The output of the sudo make install is:
Install the project...
-- Install configuration: "RelWithDebInfo"
-- Up-to-date: /usr/local/lib/mc_controller/etc/MyFirstFSMController.yaml
-- Up-to-date: /usr/local/lib/libMyFirstFSMController.so
-- Up-to-date: /usr/local/lib/mc_controller/MyFirstFSMController_controller.so
-- Up-to-date: /usr/local/lib/mc_controller/MyFirstFSMController/states/MyFirstFSMController_Initial.so
-- Up-to-date: /usr/local/lib/mc_controller/MyFirstFSMController/states/data
-- Up-to-date: /usr/local/lib/mc_controller/MyFirstFSMController/states/data/states.json

gergondet, 2021-04-26:
Hi @luigipenco 

Thank you for reporting this issue. 

How did you install mc_rtc?

Have you been able to follow other tutorials?

Can you double check the name in `src/lib.cpp`?

Could you add the following entry to your configuration (`$HOME/.config/mc_rtc/mc_rtc.yaml`) and try again?
```
VerboseLoader: true
```

Thanks

luigipenco, 2021-04-26:
Thank you for the fast reply, the other tutorials were working. By rerunning a catkin_make in my catkin_ws, mc_rtc_ticker was rebuilt and the FSM tutorial is now working.

closed 2021-04-26 by luigipenco