jrl-umi3218/mc_rtc #150: Controlling an end-effector issue

N1ckfm, 2021-04-27:
Hi, I am following this tutorial [https://jrl-umi3218.github.io/mc_rtc/tutorials/introduction/ef-controller.html#the-short-story](https://jrl-umi3218.github.io/mc_rtc/tutorials/introduction/ef-controller.html#the-short-story)
I am not sure where I should include this part of the code.

```
// Get the current objective
auto pt = efTask->get_ef_pose();
// Update the rotation and position objective
efTask->set_ef_pose(sva::PTransformd{sva::RotY(-M_PI/2), Eigen::Vector3d{0.5, -0.5, 1.2}});
```
About the task.json, I don't know in which directory it will be.
```
{
  "type": "body6d",
  "body": "l_wrist",
  "orientation": [0.71, 0, 0.71, 0],
  "position": [0.4, 0.4, 1.0]
}
```
and this:

```
// Get the task loader in there
#include <mc_tasks/MetaTaskLoader.h>

// Get the task from the JSON file
auto task = mc_tasks::MetaTaskLoader::load(solver(), "/my/path/task.json");

// Get the task from the YAML file
auto task = mc_tasks::MetaTaskLoader::load(solver(), "/my/path/task.yaml");

// Actually you can get the task from any mc_rtc::Configuration entries
auto task = mc_tasks::MetaTaskLoader::load(solver(), config("task"));

// In all the cases above, task is an std::shared_ptr<mc_tasks::MetaTask>
// but you can retrieve a more precise type, mc_rtc will check that the
// task that was retrieved from disk is compatible with the task you
// requested
auto task = mc_tasks::MetaTaskLoader::load<mc_tasks::EndEffectorTask>(solver(), config("task"));
```
I would like your support. 

Nícolas

gergondet, 2021-04-27:
Hi @N1ckfm 

> I am not sure where I should include this part of the code.

```cpp
// Get the current objective
auto pt = efTask->get_ef_pose();
// Update the rotation and position objective
efTask->set_ef_pose(sva::PTransformd{sva::RotY(-M_PI/2), Eigen::Vector3d{0.5, -0.5, 1.2}});
```

The point here is to show the API of the `EndEffectorTask`. You can use `efTask->get_ef_pose();` anywhere you would need to access the task's current target and `efTask->set_ef_pose(...)` whenever you want to change the target (for example in the reset function)

> About the task.json, I don't know in which directory it will be.

It does not matter as long as you can access this file in your code (for a simple tutorial/example you can simply hardcode the path)

> and this:

```cpp
// Get the task loader in there
#include <mc_tasks/MetaTaskLoader.h>

// Get the task from the JSON file
auto task = mc_tasks::MetaTaskLoader::load(solver(), "/my/path/task.json");

// Get the task from the YAML file
auto task = mc_tasks::MetaTaskLoader::load(solver(), "/my/path/task.yaml");

// Actually you can get the task from any mc_rtc::Configuration entries
auto task = mc_tasks::MetaTaskLoader::load(solver(), config("task"));

// In all the cases above, task is an std::shared_ptr<mc_tasks::MetaTask>
// but you can retrieve a more precise type, mc_rtc will check that the
// task that was retrieved from disk is compatible with the task you
// requested
auto task = mc_tasks::MetaTaskLoader::load<mc_tasks::EndEffectorTask>(solver(), config("task"));
```

This just shows how to use `mc_tasks::MetaTaskLoader` to get a task from a file (or a configuration entry)

Let me know if you need more explanations

N1ckfm, 2021-04-27:
Compared to the previous tutorial, the code could be copied and executed. This tutorial is not to execute a complete code.
It is only for explaining some concepts. Is it correct?

gergondet, 2021-04-28:
Yes, exactly.

That's more or less stated in the tutorial as:

> Instead of repeating a similar experience as in the previous tutorials, this tutorial will show how we can load the task from a configuration file written in JSON or YAML. In fact, this will be applicable to any task in the framework.

But perhaps this could be made clearer...

closed 2021-06-11 by gergondet