jrl-umi3218/mc_rtc #151: mc_rtc and dependencies

ssh4h, 2021-05-20:
Hi

I was wondering if I can install and use the mc_rtc platform on a system without having python installed at all. Is it possible?
If yes what parts will be missing without python? Are they just the parts related to the GUI?

Thanks,
ssh4h

gergondet, 2021-05-21:
Hi @ssh4h 

Use the mc_rtc platform without Python: yes

Install the mc_rtc platform without Python: yes as long as you specifiy `-DPYTHON_BINDING=OFF` on all dependencies

Missing parts without Python:
- No Python bindings obviously
- No [mc_log_ui](https://jrl-umi3218.github.io/mc_rtc/tutorials/tools/mc_log_ui.html)
- No `mc_rtc_new_controller` and `mc_rtc_new_fsm_controller` utils but you can the GitHub templates instead ([here](https://github.com/mc-rtc/new-controller) and [here](https://github.com/mc-rtc/new-fsm-controller))

gergondet, 2021-08-17:
Hi @ssh4h

I'm going to close this issue. Feel free to re-open if you face issues with using mc_rtc without Python on the target system.

closed 2021-08-17 by gergondet