jrl-umi3218/mc_rtc #152: Fixed-base manipulator with mc_rtc and choreonoid

mmurooka, 2021-06-02:
I would like to use some kind of fixed-base manipulator robot in the environment of mc_rtc + choreonoid.

In the [mc_rtc document](https://jrl-umi3218.github.io/mc_rtc/robots.html), Panda and Sawyer are supported for mc_rtc.

First question: where can I get the panda_description with necessary files by mc_rtc (e.g., convex collision shapes).

Second question:
We also need the choreonoid project files, python script (sim_mc.py), etc. for runnig mc_rtc on choreonoid.
For HRP robots, those files are managed in https://gite.lirmm.fr/choreonoid-hrp-installation/hrp-choreonoid.
Does anyone already have those files for fixed-base manipulator?

Third question:
We need to generate VRML robot model (*.wrl) from URDF robot model? How can I convert it?

gergondet, 2021-06-02:
Hi @mmurooka 

Everything you need is here: https://github.com/jrl-umi3218/mc_panda

It only requires `ros-${ROS_DISTRO}-libfranka` and `ros-${ROS_DISTRO}-franka-ros`. If the franka-ros meta package is not available then only franka-description should be required.

The description is based off the official panda description package from Franka Emika with just the mc_rtc related files added by mc_panda. The mc_panda project also includes Choreonoid projects that will work with mc_openrtm or mc_udp.

mmurooka, 2021-06-02:
Thank you @gergondet

Panda manipulator on mc_rtc + choreonoid worked with installing mc_panda!

Additional few questions:
- Real robot in Rviz does not reflect the actual joint positions of the robot in choreonoid. Is this expected result?
- As shown in the below video, The robot joint positions in choreonoid are bit oscillating. Is this also expeted?
https://www.dropbox.com/s/jhcjpwjuc5hmdjn/panda-choreonoid-20210602.mp4?dl=0

mmurooka, 2021-06-03:
> As shown in the below video, the robot joint positions in choreonoid are bit oscillating. Is this also expeted?

Joint oscillation stopped by setting `rotorInertia` and `gearRatio` in VRML model and tuning PDgains https://github.com/mmurooka/mc_panda/commit/26b2bb21c5ad2ee29e71a44b4e0caf67ab761ead
https://www.dropbox.com/s/72gqoe4g5m5nu85/panda-choreonoid-gain-tune-20210603.mp4?dl=0

mmurooka, 2021-06-03:
> RealRobot in Rviz does not reflect the actual joint positions of the robot in choreonoid. Is this expected result?

By adding the configurations of `ObserverPipelines` containing `Encoder` observer, the real robot state is published.

Now all questions are solved, I'll close the issue.

( If the joint oscillation is common issue, we can consider to make Pull Request of parameter updates like https://github.com/mmurooka/mc_panda/commit/26b2bb21c5ad2ee29e71a44b4e0caf67ab761ead. )

closed 2021-06-03 by mmurooka

gergondet, 2021-06-03:
Hi @mmurooka 

> If the joint oscillation is common issue

I think it is. We didn't go very far with the Choreonoid simulation as the I.AM project moved away from using it as a simulation platform. The files remained here for the sake of preservation. If you can open a pull request with your changes it would be very much appreciated. Thanks!