jrl-umi3218/mc_rtc #154: Failed in run ./build_and_install.sh, maybe RBDyn components

TsuruMasato, 2021-06-08:
Hello,

I am now updating mc_rtc.
Now I am using build_and_install.sh.

During the process,

`Building git dependency jrl-umi3218/Tasks (with test) `
`--> Compiling Tasks (branch master)`

・・・・and so on...
And finally,

`CMakeFiles/AllocationTest.dir/AllocationTest.cpp.o: 関数 `makeEnv()' 内:
/home/masato/src/Tasks/tests/arms.h:90: `rbd::MultiBodyGraph::makeMultiBody(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, sva::PTransform<double> const&, sva::PTransform<double> const&)' に対する定義されていない参照です
CMakeFiles/AllocationTest.dir/AllocationTest.cpp.o: 関数 `makeZXZArm(bool, sva::PTransform<double>)' 内:
/home/masato/src/Tasks/tests/arms.h:63: `rbd::MultiBodyGraph::makeMultiBody(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, sva::PTransform<double> const&, sva::PTransform<double> const&)' に対する定義されていない参照です
collect2: error: ld returned 1 exit status
tests/CMakeFiles/AllocationTest.dir/build.make:112: recipe for target 'tests/AllocationTest' failed
make[2]: *** [tests/AllocationTest] Error 1
make[2]: ディレクトリ '/home/masato/src/Tasks/build' から出ます
CMakeFiles/Makefile2:1489: recipe for target 'tests/CMakeFiles/AllocationTest.dir/all' failed
make[1]: *** [tests/CMakeFiles/AllocationTest.dir/all] Error 2
make[1]: *** 未完了のジョブを待っています....
CMakeFiles/TasksTest.dir/TasksTest.cpp.o: 関数 `makeEnv()' 内:
/home/masato/src/Tasks/tests/arms.h:90: `rbd::MultiBodyGraph::makeMultiBody(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, sva::PTransform<double> const&, sva::PTransform<double> const&)' に対する定義されていない参照です
CMakeFiles/TasksTest.dir/TasksTest.cpp.o: 関数 `makeZXZArm(bool, sva::PTransform<double>)' 内:
/home/masato/src/Tasks/tests/arms.h:63: `rbd::MultiBodyGraph::makeMultiBody(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, sva::PTransform<double> const&, sva::PTransform<double> const&)' に対する定義されていない参照です
`

and stoped.
Sorry, it contains some Japanese.
They are saying that there are not defined or they have not defined references.

I may have to post this issue in Tasks... I don't know well. sorry.

gergondet, 2021-06-08:
Hi @TsuruMasato 

I noticed the same issue in an incremental update. Can you run `make clean` in `Tasks/build` folder?

Then run the update script again and it should go all the way through. I am not sure why/how the build is not picking up on those RBDyn changes.

TsuruMasato, 2021-06-08:
Thank you very much for your super quick response!!
And it solved the issue.

But now I am facing another issue;

`CMakeFiles/Makefile2:4781: recipe for target 'tests/CMakeFiles/testJsonIO.dir/all' failed
make[1]: *** [tests/CMakeFiles/testJsonIO.dir/all] Error 2
../src/libmc_rbdyn.so.1.7.0: `rbd::MultiBodyGraph::makeMultiBody(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rbd::Joint::Type, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, sva::PTransform<double> const&, sva::PTransform<double> const&)' に対する定義されていない参照です
../src/libmc_rbdyn.so.1.7.0: `rbd::MultiBodyGraph::makeMultiBody(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, sva::PTransform<double> const&, sva::PTransform<double> const&)' に対する定義されていない参照です
`

It is saying that RBDyn does not have such functions...
My most concern is,  commit f52e1c96 in RBDyn, what you did on May 26th.

The function "MultiBodyGraph::makeMultiBody()" has been changed into a const function.
Maybe mc_rtc is not following this update, isn't it?
(I'm not sure.)

closed 2021-06-11 by gergondet