jrl-umi3218/mc_rtc #158: BUILD_CORE is not used in build_and_install.sh

fkanehiro, 2021-06-28:
Although it is possible to specify the number of cores by `--build-core` or `-j`, it seems not to be actually used while compliation.

gergondet, 2021-06-29:
Hi @fkanehiro 

Parallel builds are enabled via [`CMAKE_BUILD_PARALLEL_LEVEL`](https://cmake.org/cmake/help/latest/envvar/CMAKE_BUILD_PARALLEL_LEVEL.html#envvar:CMAKE_BUILD_PARALLEL_LEVEL) but unfortunately this is only available in CMake >= 3.12

You can upgrade CMake to get the functionality. I've also recently switch the script to use ninja when possible which will build in parallel regardless of the setting. However, the script won't switch existing make-based projects to ninja project.

fkanehiro, 2021-06-29:
@gergondet 
I understood what was happening yesterday. I was working with ubuntu16.04...

closed 2021-06-29 by fkanehiro