jrl-umi3218/mc_rtc #160: Simulate step-by-step / Support for Mujoco

rohanpsingh, 2021-07-05:
Hi @gergondet 

There are two parts to this issue:
1. Can mc-rtc be used to simulate a robot in Choreonoid "step-by-step"? What I want is the ability to be able to access the simulation state, apply control signals, (and also possibly change robot model parameters), and _then_ increment the simulation by one timestep. And then do this in a loop. I'm not sure if this supported by Choreonoid itself or not.

2. If it can't be done with Choreonoid, I would like to create a "new mc_rtc interface" for the Mujoco physics engine. I guess the information I need to accomplish this should be gettable [here](https://jrl-umi3218.github.io/mc_rtc/tutorials/advanced/new-interface.html), but until this page is available, could you please give me some pointers on where I could begin? And, do you think how easy/difficult would this task be?

gergondet, 2021-07-05:
Hi @rohanpsingh 

> Can mc-rtc be used to simulate a robot in Choreonoid "step-by-step"? What I want is the ability to be able to access the simulation state, apply control signals, (and also possibly change robot [...]

This is possible on mc_rtc side. In fact this is implemented in the [V-REP/CoppeliaSim interface](https://github.com/jrl-umi3218/mc_vrep) and in the ROS ticker. I'm not sure this can be implemented in Choreonoid with the current integration between mc_rtc and Choreonoid (i.e. using the OpenRTM plugin in Choreonoid). If the OpenRTM plugin can take over the simulation stepping then it would be a good start.

> If it can't be done with Choreonoid, I would like to create a "new mc_rtc interface" for the Mujoco physics engine. I guess the information I need to accomplish this should be gettable here, but [...]

Yeah. Unfortunately I didn't have the time (yet) to write the documentation of this. A good starting point is the [ticker in the mc_rtc-magnum project](https://github.com/gergondet/mc_rtc-magnum/blob/main/src/ticker/ticker.cpp). You would just need to change what happens here before/after `controller.run()` [here](https://github.com/gergondet/mc_rtc-magnum/blob/main/src/ticker/ticker.cpp#L115), i.e. read sensor data from Mujoco before running, and update the command after if `controller.run()` is successful. If you need help feel free to ping me.

rohanpsingh, 2021-07-05:
This is useful information, thanks a lot! I am a complete beginner in mc_rtc so it might take a while for me to get this done, but I will let you know here if I face issues.

rohanpsingh, 2021-09-05:
Hi @gergondet 
I have been trying to implement the interface with mujoco but am confused by a few things:  
1. What is the "required" list of sensor data that mc-rtc needs from the simulator? I saw different things that [mc_openrtm](https://github.com/jrl-umi3218/mc_openrtm/blob/master/src/MCControl.cpp#L318-L326), [mc_vrep](https://github.com/jrl-umi3218/mc_vrep/blob/main/src/vrep_simulation.cpp#L215-L256) and [magnum ticker](https://github.com/gergondet/mc_rtc-magnum/blob/main/src/ticker/ticker.cpp#L117) provide to mc_rtc. (perhaps magnum ticker can only provide encoder values since it doesn't connect to a dynamic simulator?)
2. To avoid discrepancies between our Choreonoid simulation environments and Mujoco, I want to send create direct drive actuators in the mujoco model and send directly send torque commands. Can mc_rtc provide me the toques to send using the gains defined in the robot files?
3.  Currently, I obtain the [result from QP similar to mc_openrtm](https://github.com/jrl-umi3218/mc_openrtm/blob/master/src/MCControl.cpp#L369) in the form of target positions, compute the torque myself and apply to mujoco. In this case, the robot does large random motions in mujoco and doesn't move at all in Rviz. mc_rtc gives me the following warning: 
```
[warning] [KinematicInertial] Real anchor frame jumped from [   0.240349 -0.00597792  -0.0275735] to [  0.162639   0.124982 -0.0821415] (error norm 0.000 > threshold 0.020)
[warning] [KinematicInertial] Skipping velocity update (anchor frame jumped)
```
  Perhaps you cannot explain the random motions of the robot in mujoco, but still could tell if these warnings are caused due to the same reason.

> If you need help feel free to ping me.

I could really benefit from your help here,  so please let me know if there is a better way to communicate with you :)