jrl-umi3218/mc_rtc #161: Issue with running mc_log_ui with different python versions

Ahmadreza-Shahrokhshahi, 2021-07-08:
Hello,

I am working with [LIPM walking controller](https://github.com/stephane-caron/lipm_walking_controller) (and it uses mc_rtc as in its package); I have problems using _mc_log_ui_ to open my log files. It was OK until I installed some newer versions of python to run a machine learning project. I have python2.7, python3.5, and python3.6 on my system, but now I cannot use _mc_log_ui_ with any of these python versions to open my log files. I am not sure if it is due to conflicts between these versions or there could be another reason for that.

This is for example the error I get after running _mc_log_ui_ with pyton2.7 as the default python version:
[image]

It would be appreciated if you could advise
Thanks

gergondet, 2021-07-09:
Hi @Ahmadreza-Shahrokhshahi 

Do you have a newer version of numpy of Python 2 than you had before? (It would seem so)

You can try rebuilding the Tasks bindings (since it seems to be where it fails to load for now):

To do so, go into Tasks build folder (`${SRC}/Tasks/build` assumiing `$SRC` is where all your mc_rtc related libraries are cloned and `build` is your build folder) then:
```bash
cmake --build . --target force-tasks-python-bindings
sudo cmake --build . --target install
```

You might have to repeat this if the import fails for other bindings.

Ahmadreza-Shahrokhshahi, 2021-07-16:
Thanks @gergondet for your response,
I tried to do what you suggested. To that end, I deleted the "build" folder first. Then, recreated it and ran `cmake ..` in `${SRC}/Tasks/build`, then tried what you mentioned:
`cmake --build . --target force-tasks-python-bindings`

and this was the response:
[image]

I should mention that my python and Cython versions are:
`Python 2.7.12`
`Cython version 0.29.10`

Do you have an idea what is happening?

gergondet, 2021-07-19:
Hi @Ahmadreza-Shahrokhshahi 

If you deleted the build folder you can simply run `cmake --build .` since the `force-tasks-python-bindings` target is only useful (and usable) if the bindings were built once in this build folder.

Ahmadreza-Shahrokhshahi, 2021-07-27:
It worked and I successfully ran `cmake --build .` and then `sudo cmake --build . --target install` with no error, but my main goal, which is to run a sample logfile with `mc_log_ui`is still there with the same error message!

gergondet, 2021-07-28:
Hi @Ahmadreza-Shahrokhshahi 

You mean you get the exact same error about tasks bindings?

Ahmadreza-Shahrokhshahi, 2021-07-28:
> Hi @Ahmadreza-Shahrokhshahi
> 
> You mean you get the exact same error about tasks bindings?

No, I mean the tasks buildings are done with no error, but the main issue I am looking for resolving is to run `mc_log_ui`. After building the tasks as you suggested, I came back and tried to run a log file with `mc_log_ui` (which we probably should expect to work this time), but I receive the same error that I posted at the beginning of this issue.

gergondet, 2021-07-29:
Hi @Ahmadreza-Shahrokhshahi 

Sorry if my comment was not clear. I meant to ask if you get the exact sane error message, i.e. it fails when it imports tasks and complain about a numpy issue? My expectation would be that it might fail for the same reason but in a different import (another binding would need to be rebuilt then)

A dirty option is to edit `utils/mc_log_gui/mc_log_ui/mc_log_ui.py` in mc_rtc and at line 30 replace `except ImportError:` with `except:`

Ahmadreza-Shahrokhshahi, 2021-07-29:
Hi @gergondet ,

Finally, this last suggestion made the `mc_log_ui` work and now I can successfully open my log files.
Thank you very much

commit 1c206bc4 (2020-02-05): Merge branch 'topic/ConstraintDocumentation' into 'master'

[doc] Improve documentation for *EqualityConstraint helper

Closes #161

See merge request multi-contact/mc_rtc!194
commit f7b46af6 (2020-02-05): [doc] Improve documentation for *EqualityConstraint helper

- Hides helper classes in the utils namespace
- Use a single constructor for specialized constraints
- Documentat specialized constraints

Closes #161
closed 2021-07-29 by Ahmadreza-Shahrokhshahi