jrl-umi3218/mc_rtc #176: mc_rtc_ticker version 1.6.0-ccb1b8567 / RviZ not working after build

Sil87, 2021-07-27:
Hello, 
I did yesterday a  ./build_and_install.sh and RviZ  stopped working (screen capture attached)
The error I get: 
" mc_rtc_ticker was compiled with 1.6.0-ccb1b8567 but mc_rtc is at version 1.6.0-ccb1b8567-dirty, you might face subtle issues or unexpected crashes, please recompile mc_rtc_ticker"

Plus some warnings related to libraries and default controllers when i run mc_rtc_ticker
[image]
[image]

Do you have any advice for me? Thank you, 
Silvia

Sil87, 2021-07-27:
Also did what was suggested in issue  #138. Deleted the build folder and run a cmake/ sudo make intall same error and warning messages.

gergondet, 2021-07-28:
Hi @Sil87 

Does it work with the sample controllers? (e.g `Posture`) If so, please recompile your controller and try again.

If not, do you see this line when you start the ticker:
```
[success] [mc_rtc::ROS] Starting ROS services
```

Best regards,

Sil87, 2021-07-28:
Hello @gergondet 
Not working with sample controllers, when i start the ticker i don't see the ROS message, however I do seem to be connected to ROS according to print screen: 

[image]

At the end of the warnings,  I also get: 
[info] MCController(base) ready
munmap_chunk(): invalid pointer
Aborted (core dumped)

gergondet, 2021-07-28:
Hi @Sil87 

> At the end of the warnings, I also get:
[info] MCController(base) ready
munmap_chunk(): invalid pointer
Aborted (core dumped)

You get this with a sample controller? If so, I suggest you rebuild mc_rtc_ros:
```bash
cd ~/MCRTC/catkin_ws
catkin_make clean
catkin_make
```

Sil87, 2021-07-28:
Ok thank you, will try that. I might have done some "faux pas" some weeks ago when I deleted the folder of an old controller i created that kept giving me errors.

Sil87, 2021-07-28:
So I got this error after clean:
[ERROR] [1627468565.020266690]: PluginlibFactory: The plugin for class 'mc_rtc_rviz_panel/mc_rtc panel' failed to load.  Error: Could not find library corresponding to plugin mc_rtc_rviz_panel/mc_rtc panel. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.

And this one after catkin_make:
The specified source space "/home/silvia/src" does not exist

gergondet, 2021-07-29:
This particular issue was fixed. The problem was combining a package-based installation (out-dated because it was using the bintray mirror) with a source-based installation. This has been fixed but the script should detect such problematic situations that will be tracked in #179

closed 2021-07-29 by gergondet