jrl-umi3218/mc_rtc #177: Robot falls down with StabilizerTask when standing far from the origin

mmurooka, 2021-07-28:
Using lipm_walking_controller of [topic/stabilizer_ana branch](https://github.com/arntanguy/lipm_walking_controller/tree/topic/stabilizer_ana) with https://github.com/mmurooka/lipm_walking_controller/commit/eb9c91e541f8dafac5586324c844625b37d6a0e2 patch to set the robot initial standing pose, the robot falls down when StabilizerTask is added.

I confirmed that the same issue happens for JVRC1, HRP2KAI, HRP5P simulation.
https://www.dropbox.com/s/mkspmq4hh6bb868/stabilizer-initial-pos-non-origin-20210728.mp4?dl=0

mitsuharu-morisawa, 2021-07-29:
When I tested by DCMStabilizer modified from StabilizerTask [branch is topic/DCMStabilizer](https://github.com/mitsuharu-morisawa/mc_rtc/tree/topic/DCMStabilizer), the problem was not happen.
LIPM is also adapted to DCMStabilizer in this repo [branch is topic/dcm_stabilizer_ana](https://github.com/mitsuharu-morisawa/lipm_walking_controller/tree/topic).
I confirmed this by HRP2KAI and HRP5P. Both were not happen.

So I think the problem is around force distribution or dcm controller in mc_rtc's StabilizerTask.

rohanpsingh, 2022-06-15:
I faced the same problem with HRP5P with lipm_walking_controller in mujoco.
(cc: @mehdi-benallegue )

mitsuharu-morisawa, 2022-06-15:
Can you try to increase P gain of DCM in StabilizerTask ?

rohanpsingh, 2022-06-16:
@mitsuharu-morisawa @mehdi-benallegue 
I can reproduce this in Choreonoid too.

arntanguy, 2022-06-20:
I can also reproduce this behaviour. I'll try to investigate ASAP, I couldn't find an obvious cause when I looked at it last thursday.

mmurooka, 2022-09-02:
`init_pos` is not working in choreonoid as being tracked in https://github.com/jrl-umi3218/mc_rtc/issues/284.
If you want to reproduce this robot fall for the time being, you can change the lipm_walking_controller code as follows: https://github.com/mmurooka/lipm_walking_controller/commit/c9a6f30970823ebf8ff06c5930df58c68de4121d

arntanguy, 2022-09-08:
Should have been fixed by #285. If you notice this issue arising again during experiments, feel free to reopen.

closed 2022-09-08 by arntanguy