jrl-umi3218/mc_rtc #182: Conversion from own type to mc_rtc::configuration

mmurooka, 2021-08-06:
I have a question about conversion from own type instance to mc_rtc::configuration.

For the class `MyType` where [`ConfigurationLoader::load` and `ConfigurationLoader::save`](https://jrl-umi3218.github.io/mc_rtc/tutorials/usage/mc_rtc_configuration.html#support-for-your-own-types
) methods are defined, how can I convert `MyType` instance to the configuration?

I have confirmed that conversion is possible via the dummy configuration and key (empty string) as shown below, but is it possible to convert directly without using these dummy ones?
```
mc_rtc::Configuration convertToConfiguration(const MyType & obj)
{
  mc_rtc::Configuration tmp_config;
  tmp_config.add("", obj);
  return tmp_config("");
}
```

gergondet, 2021-08-06:
Hi @mmurooka 

I think the following should do the trick:
```cpp
auto converted = mc_rtc::ConfigurationLoader<MyType>::save(obj);
```

For a snappier version:
```cpp
inline mc_rtc::Configuration to_config(const MyType & obj)
{
  return mc_rtc::ConfigurationLoader<MyType>::save(obj);
}

// alt as a method in MyType
mc_rtc::Configuration MyType::to_config() const
{
  return mc_rtc::ConfigurationLoader<MyType>::save(*this);
}
```

mmurooka, 2021-08-06:
It worked! Thank you @gergondet

closed 2021-08-06 by mmurooka