jrl-umi3218/mc_rtc #183: Install failed in conda environment

rohanpsingh, 2021-08-11:
@gergondet Sorry to bother you.

I am trying to install mc_rtc in a conda virtual environment with `export PYTHON_BUILD_PYTHON2_AND_PYTHON3="true"`,  
and the script fails while trying to build SpaceVelAlg.

Steps that lead to this error:
```
$ cmake /home/rohan/src/SpaceVecAlg -DCMAKE_INSTALL_PREFIX:STRING=/usr/local -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=ON -DPYTHON_BINDING:BOOL=ON -DPYTHON_BINDING_USER_INSTALL:BOOL=OFF -DPYTHON_BINDING_FORCE_PYTHON2:BOOL=OFF -DPYTHON_BINDING_FORCE_PYTHON3:BOOL=OFF -DPYTHON_BINDING_BUILD_PYTHON2_AND_PYTHON3:BOOL=ON -DMC_LOG_UI_PYTHON_EXECUTABLE:STRING=python -DCMAKE_BUILD_TYPE:STRING=RelWithDebInfo -GNinja -DBUILD_PYTHON_INTERFACE:BOOL=OFF -DSPDLOG_BUILD_EXAMPLE:BOOL=OFF -DSPDLOG_BUILD_SHARED:BOOL=ON
$ cmake --build . --config RelWithDebInfo
```
This will fail with the following:
```
[12/12] Generating local sva python3 bindings
FAILED: binding/python/CMakeFiles/sva-python3-bindings 
cd /home/rohan/src/SpaceVecAlg/build/binding/python && /usr/bin/cmake -E chdir /home/rohan/src/SpaceVecAlg/build/binding/python/sva/python3/RelWithDebInfo python3 setup.py build_ext --inplace
/home/rohan/anaconda3/envs/mcrtc/lib/python3.6/site-packages/Cython/Compiler/Main.py:369: FutureWarning: Cython directive 'language_level' not set, using 2 for now (Py2). This will change in a later release! File: /home/rohan/src/SpaceVecAlg/build/binding/python/sva/python3/RelWithDebInfo/sva/sva.pxd
  tree = Parsing.p_module(s, pxd, full_module_name)

Error compiling Cython file:
------------------------------------------------------------
...
#
# Copyright 2012-2019 CNRS-UM LIRMM, CNRS-AIST JRL
#

cimport eigen.eigen as eigen
       ^
------------------------------------------------------------

sva/sva.pxd:5:8: 'eigen/eigen.pxd' not found

Error compiling Cython file:
------------------------------------------------------------
...
#
# Copyright 2012-2019 CNRS-UM LIRMM, CNRS-AIST JRL
#

cimport eigen.c_eigen as c_eigen
       ^
------------------------------------------------------------

```

gergondet, 2021-08-11:
> But now fails in building the catkin workspaces. Let me investigate further.

Could you elaborate? Does it fail when building the `catkin_data_ws` or the `catkin_ws`? It might be because there's a clash between conda Python and ROS Python.

If it fails on building the data workspace, it doesn't matter too much because there's no code that we rely on. You can comment line 1212 in build_and_install.sh and it should work to build mc_rtc in the conda workspace (you might want to comment line 1323 as well to skip thhe other workspace build)

rohanpsingh, 2021-08-11:
>  It might be because there's a clash between conda Python and ROS Python.

It does seem so. Originally, the error was this:
```
####
#### Running command: "cmake /home/rohan/src/catkin_data_ws/src -DCATKIN_DEVEL_PREFIX=/home/rohan/src/catkin_data_ws/devel -DCMAKE_INSTALL_PREFIX=/home/rohan/src/catkin_data_ws/install -G Unix Makefiles" in "/home/rohan/src/catkin_data_ws/build"
####
CMake Error at CMakeLists.txt:22 (message):
  Search for 'catkin' in workspace failed (catkin_find_pkg catkin
  /home/rohan/src/catkin_data_ws/src): Traceback (most recent call last):

    File "/usr/bin/catkin_find_pkg", line 6, in <module>
      from pkg_resources import load_entry_point
    File "/home/rohan/anaconda3/envs/mcrtc/lib/python3.6/site-packages/pkg_resources/__init__.py", line 1367
      raise SyntaxError(e) from e
                              ^

  SyntaxError: invalid syntax

-- Configuring incomplete, errors occurred!
See also "/home/rohan/src/catkin_data_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/rohan/src/catkin_data_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
File "/home/rohan/src/catkin_ws/src/CMakeLists.txt" already existsBase path: /home/rohan/src/catkin_ws
Source space: /home/rohan/src/catkin_ws/src
Build space: /home/rohan/src/catkin_ws/build
Devel space: /home/rohan/src/catkin_ws/devel
Install space: /home/rohan/src/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/rohan/src/catkin_ws/build"
####
CMake Error at CMakeLists.txt:22 (message):
  Search for 'catkin' in workspace failed (catkin_find_pkg catkin
  /home/rohan/src/catkin_ws/src): Traceback (most recent call last):

    File "/usr/bin/catkin_find_pkg", line 6, in <module>
      from pkg_resources import load_entry_point
    File "/home/rohan/anaconda3/envs/mcrtc/lib/python3.6/site-packages/pkg_resources/__init__.py", line 1367
      raise SyntaxError(e) from e
                              ^

  SyntaxError: invalid syntax

-- Configuring incomplete, errors occurred!
See also "/home/rohan/src/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/rohan/src/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:2518: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
````

Now I am trying with line 1212 commented out.

> If it fails on building the data workspace, it doesn't matter too much because there's no code that we rely on. 

Really?

gergondet, 2021-08-11:
> It does seem so. Originally, the error was this:

Yup. Looks like there's no catkin in your conda environment (of course) and so it fails to build.

> If it fails on building the data workspace, it doesn't matter too much because there's no code that we rely on.

Yes. The `catkin_data_ws` contains only data (typically, `mc_rtc_data` and some robots' description packages) so you don't need to build anything specific for the conda workspace and mc_rtc should still be able to find the data.

rohanpsingh, 2021-08-11:
As far as I understood I needed to comment out the whole block from 1210 to 1215. I did.

Now, it seems mc_rtc build can't find `mc_env_description`, for example :sob: 

```
CMake Error at CMakeLists.txt:112 (find_package):
  By not providing "Findmc_env_description.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "mc_env_description", but CMake did not find one.

  Could not find a package configuration file provided by
  "mc_env_description" with any of the following names:

    mc_env_descriptionConfig.cmake
    mc_env_description-config.cmake

  Add the installation prefix of "mc_env_description" to CMAKE_PREFIX_PATH or
  set "mc_env_description_DIR" to a directory containing one of the above
  files.  If "mc_env_description" provides a separate development package or
  SDK, be sure it has been installed.
```

gergondet, 2021-08-11:
Did you `source catkin_data_ws/devel/setup.bash` in your environment?

Basically, you should have something similar to this (assuming you have another installation of mc_rtc):
```bash
❯ echo $CMAKE_PREFIX_PATH
/home/gergondet/devel/src/catkin_ws/devel:/home/gergondet/devel/src/catkin_data_ws/devel:/opt/ros/melodic
```

rohanpsingh, 2021-08-11:
Solved.  
I needed to build the catkin workspaces outside the conda environment, because I won't be able to build them in the python3.6 environment, as expected. Then, I need to execute build_and_install again in the inside the conda environment. But this time, I must set `BUILD_SUBDIR` and `INSTALL_PREFIX` like you mentioned https://github.com/jrl-umi3218/mc_rtc/issues/183#issuecomment-895934763 .  And of course comment out the lines for building both the catkin workspaces.  

Thanks a lot for your help Pierre! :pray:

commit 0f47a0a8 (2020-01-28): [samples] Make FSM into its own sample

- Stop with an error message if ran with an unsupported robot (only JVRC1 supported)

Closes #183
commit 95beef36 (2020-01-28): Merge branch 'topic/test_samples' into 'master'

Test samples and improve FSM sample controller

Closes #183

See merge request multi-contact/mc_rtc!201
closed 2021-08-11 by rohanpsingh